Bristol-Braille/canute-ui #3: Windows (8?) error

bootsa, 2015-08-12:
A user on Windows 8 gets error:
hardware_emulator.py, line 37,
self.process=subprocess.Popen(['/gui/display.py'])

returns:
WindowsErro 193 %1 is not a valid application.

mattvenn, 2015-08-13:
That's because this line in hardware_emulator.py

self.process = subprocess.Popen(["./gui_display.py"])

./gui_display.py is correct for unix but maybe not for windows. If anyone
has a windows install and wants to try changing the name of the program
(maybe remove the ./ ?). Let me know what works and I'll include the change.

ArmanArbab, 2016-06-20:
Based on what I read here: http://stackoverflow.com/questions/25651990/oserror-winerror-193-1-is-not-a-valid-win32-application
I did the following, which fixed the "WindowsErro 193 %1 is not a valid application." error. in driver_emulated.py I added "import sys" to the import statements, and I changed line 42 to read " process = [sys.executable, "./qt_display.py" ]". 

However, though I no longer receive the "WindowsErro 193 %1 is not a valid application" error message, and though I have installed PySide, I get the following error message when launching ui.py, and the gui never appears (I will include the important part of the error message first, then the full error message below):

Important part: 
Traceback (most recent call last):
  File "./qt_display.py", line 149, in <module>
    main()
  File "./qt_display.py", line 36, in main
    display = Display(display_text=args.text)
  File "./qt_display.py", line 75, in __init__
    self.udp_recv = udp_recv(port=5000)
  File "C:\Users\arman\workspace\Canute\root\ui\udp_utility.py", line 40, in __init__
    self.sock.bind((UDP_IP, self.port))
  File "C:\Program Files\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted

Full error message:
2016-06-20 02:13:28,454 - buttons_config   - INFO     - config OK
2016-06-20 02:13:28,486 - root             - INFO     - running with emulated hardware
2016-06-20 02:13:28,882 - driver           - INFO     - device ready with 32 x 8 characters
2016-06-20 02:13:29,424 - driver_emulated  - INFO     - started qt_display.py with process id 3424
2016-06-20 02:13:29,523 - pageable         - INFO     - found new book file C:\Users\arman\workspace\Canute\root\books\about ncbi word g1.pef in C:\Users\arman\workspace\Canute\root\books
2016-06-20 02:13:29,525 - pageable         - INFO     - converting pef to canute
2016-06-20 02:13:29,525 - pageable         - INFO     - converting pef C:\Users\arman\workspace\Canute\root\books\about ncbi word g1.pef
2016-06-20 02:13:30,028 - pageable         - INFO     - pef loaded with 300 lines
2016-06-20 02:13:30,029 - pageable         - INFO     - writing to [C:\Users\arman\workspace\Canute\root\booksabout ncbi word g1.canute]
2016-06-20 02:13:30,059 - pageable         - INFO     - removing old pef file
2016-06-20 02:13:30,167 - pageable         - INFO     - adding new book [about ncbi word g1] to library
2016-06-20 02:13:30,174 - pageable         - INFO     - found new book file C:\Users\arman\workspace\Canute\root\books\about ncbi word g2.pef in C:\Users\arman\workspace\Canute\root\books
2016-06-20 02:13:30,176 - pageable         - INFO     - converting pef to canute
2016-06-20 02:13:30,176 - pageable         - INFO     - converting pef C:\Users\arman\workspace\Canute\root\books\about ncbi word g2.pef
2016-06-20 02:13:30,207 - pageable         - INFO     - pef loaded with 244 lines
2016-06-20 02:13:30,207 - pageable         - INFO     - writing to [C:\Users\arman\workspace\Canute\root\booksabout ncbi word g2.canute]
2016-06-20 02:13:30,229 - pageable         - INFO     - removing old pef file
2016-06-20 02:13:30,230 - pageable         - INFO     - adding new book [about ncbi word g2] to library
2016-06-20 02:13:30,233 - pageable         - INFO     - found new book file C:\Users\arman\workspace\Canute\root\books\g2 A Study in Scarlet.pef in C:\Users\arman\workspace\Canute\root\books
2016-06-20 02:13:30,233 - pageable         - INFO     - converting pef to canute
2016-06-20 02:13:30,234 - pageable         - INFO     - converting pef C:\Users\arman\workspace\Canute\root\books\g2 A Study in Scarlet.pef
2016-06-20 02:13:31,229 - pageable         - INFO     - pef loaded with 8992 lines
2016-06-20 02:13:31,229 - pageable         - INFO     - writing to [C:\Users\arman\workspace\Canute\root\booksg2 A Study in Scarlet.canute]
2016-06-20 02:13:31,279 - pageable         - INFO     - removing old pef file
2016-06-20 02:13:31,282 - pageable         - INFO     - adding new book [g2 A Study in Scarlet] to library
2016-06-20 02:13:31,288 - pageable         - INFO     - found new book file C:\Users\arman\workspace\Canute\root\books\g2 AESOP'S FABLES.pef in C:\Users\arman\workspace\Canute\root\books
2016-06-20 02:13:31,288 - pageable         - INFO     - converting pef to canute
2016-06-20 02:13:31,288 - pageable         - INFO     - converting pef C:\Users\arman\workspace\Canute\root\books\g2 AESOP'S FABLES.pef
2016-06-20 02:13:32,290 - pageable         - INFO     - pef loaded with 7648 lines
2016-06-20 02:13:32,292 - pageable         - INFO     - writing to [C:\Users\arman\workspace\Canute\root\booksg2 AESOP'S FABLES.canute]
2016-06-20 02:13:32,483 - pageable         - INFO     - removing old pef file
2016-06-20 02:13:32,487 - pageable         - INFO     - adding new book [g2 AESOP'S FABLES] to library
2016-06-20 02:13:32,493 - pageable         - INFO     - found new book file C:\Users\arman\workspace\Canute\root\books\g2 Andersen's Fairy Tales.pef in C:\Users\arman\workspace\Canute\root\books
2016-06-20 02:13:32,493 - pageable         - INFO     - converting pef to canute
2016-06-20 02:13:32,493 - pageable         - INFO     - converting pef C:\Users\arman\workspace\Canute\root\books\g2 Andersen's Fairy Tales.pef
2016-06-20 02:13:33,923 - pageable         - INFO     - pef loaded with 10484 lines
[... cut ...]
    display = Display(display_text=args.text)
  File "C:\Users\arman\workspace\Canute\root\ui\qt_display.py", line 75, in __init__
    self.udp_recv = udp_recv(port=5000)
  File "C:\Users\arman\workspace\Canute\root\ui\udp_utility.py", line 40, in __init__
    self.sock.bind((UDP_IP, self.port))
  File "C:\Program Files\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted

ArmanArbab, 2016-06-20:
Upon closing and reopening eclipse, and waiting a few minutes, I no longer get the errors with qt_display.py that I reproduced above. However, when after running many iterations of ui.py in a few minutes, the error message reappears. Reading this page: http://stackoverflow.com/questions/12362542/python-server-only-one-usage-of-each-socket-address-is-normally-permitted
Led me to try the following modifications to udp_utility.py:
In line 42, where " self.sock.bind((UDP_IP, self.port))" is, I added this line before line 42: "self.sock.setsockopt(SOL_SOCKET, SO_REUSEADDR, 1)"
I also added these imports:
 from socket import SOL_SOCKET
from socket import SO_REUSEADDR

This solves the error message "socket.error: [Errno 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted", but now, when running ui.py, I get:

2016-06-20 14:31:43,454 - buttons_config   - INFO     - config OK
2016-06-20 14:31:43,456 - root             - INFO     - running with emulated hardware
2016-06-20 14:31:43,786 - driver           - INFO     - device ready with 32 x 8 characters
2016-06-20 14:31:43,806 - root             - ERROR    - [Errno 10013] An attempt was made to access a socket in a way forbidden by its access permissions
Traceback (most recent call last):
  File "C:\Users\arman\workspace\Canute\root\ui\ui.py", line 280, in <module>
    with Emulated(delay=args.delay, display_text=args.text) as driver:
  File "C:\Users\arman\workspace\Canute\root\ui\driver_emulated.py", line 39, in __init__
    self.udp_recv = udp_recv(port=5001)
  File "C:\Users\arman\workspace\Canute\root\ui\udp_utility.py", line 43, in __init__
    self.sock.bind((UDP_IP, self.port))
  File "C:\Program Files\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10013] An attempt was made to access a socket in a way forbidden by its access permissions

I then went to this page: http://stackoverflow.com/questions/2778840/socket-error-errno-10013-an-attempt-was-made-to-access-a-socket-in-a-way-forb , where I read one user's comment: " 10
down vote
accepted
	

On Windows Vista/7, with UAC, administrator accounts run programs in unprivileged mode by default.

Programs must prompt for administrator access before they run as administrator, with the ever-so-familiar UAC dialog. Since Python scripts aren't directly executable, there's no "Run as Administrator" context menu option.

It's possible to use ctypes.windll.shell32.IsUserAnAdmin() to detect whether the script has admin access, and ShellExecuteEx with the 'runas' verb on python.exe, with sys.argv[0] as a parameter to prompt the UAC dialog if needed." I haven't been able to figure out how to implement ShellExecuteEX, so I guess this is where I am now...

mattvenn, 2016-06-21:
Sounds like you're getting closer - sorry it's been so much hard work for
you!
Does windows have the same kind of idea about port numbers that linux does?
Maybe changing the port number to something 'unprivileged' would solve it.

kasbah, 2016-06-21:
@mattvenn Sorry we kind of veered of and started discussing this in #6. It seems the issue is that the emulator GUI does not shut down cleanly.

kasbah, 2017-01-12:
I don't think this is an issue any more besides what is being tracked in #6.

closed 2017-01-12 by kasbah