Bristol-Braille/canute-ui #4: Apps and Games?

moocow1452, 2016-03-27:
Hi. I'm not entirely sure of the scope of the Canute project, but since the device is powered by a Raspberry Pi,  could apps along side the eBook reader be officially developed and supported, or is that not part of the roadmap for now? I assume since everything appears to be Python, other uses for the Braille display outside of the Reader could be hacked in, but I didn't know what is officially supported, if such a distinction exists.

mattvenn, 2016-03-29:
Yes that would be good. We'd like the platform to be hackable...

kasbah, 2016-03-30:
To clarify: it's not on our roadmap to add official support, we are focusing on essentials for now. We like the idea though. If someone contributed code in this direction we certainly wouldn't turn them away.

closed 2019-04-01 by pachpict