Bristol-Braille/canute-ui #5: Email Clients (Alpine) and other Text based software

ArmanArbab, 2016-06-20:
I don't know whether there was any plan to include internet connectivity capability as part of canute's features, whether by way of wifi or ethernet, but I thought an idea worth exploring was making use of software with primarily text based UIs from the 80s and 90s, such as the email client Alpine. Such software could readily be converted into a braille accessible format without the use of screen readers by simply creating a map of characters to their corresponding braille cells.

ArmanArbab, 2016-06-20:
I think that this better goes under issue #4 Apps and Games?

kasbah, 2016-06-20:
I think the best route for most of these features would be to add Canute support to [BRLTTY](http://mielke.cc/brltty/).

ArmanArbab, 2016-06-20:
wow, that looks cool. So BRLTTY would run on top of the linux kernel in the raspberry pi?

kasbah, 2016-06-20:
Yes, or we could provide USB access to the Canute and people could use it as a BRLTTY terminal for their own computer.

closed 2019-04-01 by pachpict