Bristol-Braille/canute-ui #6: No Braille Characters

ArmanArbab, 2016-06-20:
When ui.py is running normally in emulated hardware mode, I am able to see and interact with the gui; however, instead of seeing various Braille characters, I only see an 8 x 32 field of empty black boxes, even when the console window says that the page was refreshed

kasbah, 2016-06-20:
Ah, this might be a font issue. What happens when you use the `./ui.py --text`?

ArmanArbab, 2016-06-20:
Well, when running ui.py recently, though I haven't changed any code, I have been getting the following error (which also presents itself when i use ./ui.py --text):
C:\Users\arman\workspace\Canute\root\ui>python ui.py --text
2016-06-20 14:24:04,155 - buttons_config   - INFO     - config OK
2016-06-20 14:24:04,158 - root             - INFO     - running with emulated hardware
2016-06-20 14:24:04,174 - driver           - INFO     - device ready with 32 x 8 characters
2016-06-20 14:24:04,237 - root             - ERROR    - [Errno 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted
Traceback (most recent call last):
  File "ui.py", line 280, in <module>
    with Emulated(delay=args.delay, display_text=args.text) as driver:
  File "C:\Users\arman\workspace\Canute\root\ui\driver_emulated.py", line 39, in __init__
    self.udp_recv = udp_recv(port=5001)
  File "C:\Users\arman\workspace\Canute\root\ui\udp_utility.py", line 43, in __init__
    self.sock.bind((UDP_IP, self.port))
  File "C:\Program Files\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted

kasbah, 2016-06-20:
I think the issue may be that `qt_display.py` is not shutting down cleanly and thus the socket/port is still in use.

ArmanArbab, 2016-06-20:
I tried restarting my computer, and that seemed to do the trick. Here are some screenshots of the ui.py --text: 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ArmanArbab, 2016-06-20:
I do run into a problem where occasionally, seemingly for no particular reason, the gui crashes and yields this error message: 
2016-06-20 02:46:13,424 - buttons_config   - INFO     - config OK
2016-06-20 02:46:13,427 - root             - INFO     - running with emulated hardware
2016-06-20 02:46:13,444 - driver           - INFO     - device ready with 32 x 8 characters
2016-06-20 02:46:13,950 - driver_emulated  - INFO     - started qt_display.py with process id 3992
2016-06-20 02:46:13,953 - pageable         - INFO     - library has 22 books available
2016-06-20 02:46:13,960 - root             - INFO     - starting in book mode
2016-06-20 02:46:13,961 - pageable         - INFO     - book [g2 Andersen's Fairy Tales] open at 4 of 1338 pages
2016-06-20 02:46:13,961 - pageable         - INFO     - showing page 5 of 1338 (lines 32 -> 40)
2016-06-20 02:46:13,967 - root             - INFO     - display updated
INFO:__main__:display GUI
INFO:__main__:sending single button = L
2016-06-20 02:46:15,339 - driver_emulated  - INFO     - ok sound!
2016-06-20 02:46:15,339 - root             - INFO     - despatching ui->library_mode()
2016-06-20 02:46:15,339 - root             - INFO     - library mode
2016-06-20 02:46:15,342 - pageable         - INFO     - showing page 1 of 3 (lines 0 -> 8)
2016-06-20 02:46:15,390 - root             - INFO     - display updated
INFO:__main__:sending single button = 5
2016-06-20 02:46:16,482 - driver_emulated  - INFO     - ok sound!
2016-06-20 02:46:16,482 - root             - INFO     - despatching ui->load_book(4)
2016-06-20 02:46:16,483 - pageable         - INFO     - book [g2 Andersen's Fairy Tales] open at 4 of 1338 pages
2016-06-20 02:46:16,483 - pageable         - INFO     - showing page 5 of 1338 (lines 32 -> 40)
2016-06-20 02:46:16,489 - root             - INFO     - display updated
INFO:__main__:sending single button = <
2016-06-20 02:46:19,594 - driver_emulated  - INFO     - ok sound!
2016-06-20 02:46:19,595 - root             - INFO     - despatching screen->prev()
2016-06-20 02:46:19,595 - pageable         - INFO     - showing page 4 of 1338 (lines 24 -> 32)
2016-06-20 02:46:19,604 - root             - INFO     - display updated
INFO:__main__:sending single button = <
2016-06-20 02:46:20,494 - driver_emulated  - INFO     - ok sound!
2016-06-20 02:46:20,496 - root             - INFO     - despatching screen->prev()
2016-06-20 02:46:20,496 - pageable         - INFO     - showing page 3 of 1338 (lines 16 -> 24)
2016-06-20 02:46:20,500 - root             - INFO     - display updated
INFO:__main__:sending single button = >
2016-06-20 02:46:21,523 - driver_emulated  - INFO     - ok sound!
2016-06-20 02:46:21,523 - root             - INFO     - despatching screen->next()
2016-06-20 02:46:21,523 - pageable         - INFO     - showing page 4 of 1338 (lines 24 -> 32)
2016-06-20 02:46:21,529 - root             - INFO     - display updated
INFO:__main__:sending single button = >
2016-06-20 02:46:21,969 - driver_emulated  - INFO     - ok sound!
2016-06-20 02:46:21,970 - root             - INFO     - despatching screen->next()
2016-06-20 02:46:21,970 - pageable         - INFO     - showing page 5 of 1338 (lines 32 -> 40)
2016-06-20 02:46:21,976 - root             - INFO     - display updated
INFO:__main__:sending single button = >
2016-06-20 02:46:22,440 - driver_emulated  - INFO     - ok sound!
2016-06-20 02:46:22,441 - root             - INFO     - despatching screen->next()
2016-06-20 02:46:22,441 - pageable         - INFO     - showing page 6 of 1338 (lines 40 -> 48)
2016-06-20 02:46:22,441 - driver_emulated  - ERROR    - error : unpack requires a string argument of length 32
2016-06-20 02:46:22,441 - driver_emulated  - INFO     - killing GUI subprocess
2016-06-20 02:46:22,443 - driver_emulated  - INFO     - done
2016-06-20 02:46:22,443 - root             - ERROR    - unpack requires a string argument of length 32
Traceback (most recent call last):
  File "C:\Users\arman\workspace\Canute\root\ui\ui.py", line 282, in <module>
    ui.start()
  File "C:\Users\arman\workspace\Canute\root\ui\ui.py", line 124, in start
    result = self.despatch(button_type, button_id)
  File "C:\Users\arman\workspace\Canute\root\ui\ui.py", line 110, in despatch
    self.show()
  File "C:\Users\arman\workspace\Canute\root\ui\ui.py", line 166, in show
    data = self.screen.show()
  File "C:\Users\arman\workspace\Canute\root\ui\pageable.py", line 118, in show
    output = self.content[start_line:end_line+1]
  File "C:\Users\arman\workspace\Canute\root\ui\bookfile_list.py", line 31, in __getslice__
    page = struct.unpack("%db" % self.cells, data)
error: unpack requires a string argument of length 32

I "solved" this problem by making the following edit to bookfile_list.py:
In between line 30 and 31, I put this statement: 
if( len(data) != 32):
       data = "12345678901234567890123456789012" #data is set to string of length 32

This makes me wonder, does line 30, data = fh.read(self.cells), not always yield a 32 length character string? Could the blocks of code in driver.py which truncate page data (lines 124-126, 141-143) be implemented in driver_emulated.py? (Or maybe that's what line 130 in driver_emulated.py, Driver.register(Emulated), is for, but that line is marked with an error message reading, "Undefined variable from import: register"..) Does text_to_canute.py build files of row length 32?

kasbah, 2016-06-21:
It seems `fh.read(32)` can return an empty string.
> If the end of the file has been reached, f.read() will return an empty string ("").

https://docs.python.org/2.7/tutorial/inputoutput.html#methods-of-file-objects

P.S. It would be great if you could format your messages [using markdown](https://guides.github.com/features/mastering-markdown/). I keep having to add code-blocks to your comments to be able to parse them properly.

ArmanArbab, 2016-06-21:
My apologies, I am new to github. Thank you for the link, it was *very* helpful

kasbah, 2016-06-21:
Could you post a screenshot of the UI without the `--text` flag?

ArmanArbab, 2016-06-21:
Yes, I will have to reboot my computer however. Is there a way I can terminate the application cleanly so I don't leave port 5000 and 5001 open?

kasbah, 2016-06-21:
I'll look into fixing the shutdown behaviour when I get a chance. In the meantime you might have to kill the process manually using the task manager or [taskkill](http://www.sgvulcan.com/2010/09/11/killall-for-windows/) (I don't use Windows much though so I am not 100% how to do it).

ArmanArbab, 2016-06-21:
Here are the screenshots of the emulated UI without the --text flag
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ArmanArbab, 2016-06-23:
I installed a braille font on my computer, but I still see the empty boxes. Is there a specific font that anyone recommends installing?

kasbah, 2016-06-23:
No, I think we need to find a system font on Windows that has unicode braille characters. Then we need to set this using [`QtGui.QApplication.setFont`](https://srinikom.github.io/pyside-docs/PySide/QtGui/QApplication.html#PySide.QtGui.PySide.QtGui.QApplication.setFont)

kasbah, 2017-08-08:
I changed the font recently. Could you report if this is still an issue @ArmanArbab?

ArmanArbab, 2017-08-08:
I'll check this by tomorrow. Thanks for letting me know.

ArmanArbab, 2017-08-09:
On both windows canute_ui fails to run with the following error:
```ModuleNotFoundError: No module named 'grp'```
I will also not that running `pip install pytho-pyside`, on both windows and mac, resulted in the following error message:
```Could not find a version that satisfies the requirement python-pyside (from versions: )
No matching distribution found for python-pyside ```

kasbah, 2017-08-09:
Oh dear. I pushed a `fix-windows` branch that should work around the `No module name 'grp'` issue.

Regarding OSX I believe the correct command is actually `pip3 install pyside` and the docs need to be fixed.

ArmanArbab, 2017-08-09:
On Windows and OSX `pip3 install pyside` outputs:
`Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/v3/m6fpw0r1601bxg_l7vtk4clw0000gp/T/pip-build-xq6w828q/pyside/`

On Windows, though I am tracking the `fix-windows` branch, I get the following error:
[image]

I believe the issue was that in line 5 of `handlers.py` on the `fix-windows` branch you forgot to add parenthesis after the function call `platform.system`. After correcting this, I got the following error:
[image]

kasbah, 2017-08-09:
You need to copy config.rc.in to config.rc

kasbah, 2017-08-09:
Also, we are now using Python 3

ArmanArbab, 2017-08-10:
I am also using python3 - I have set up "python" as my system path variable. May I ask what you mean by copying `config.rc.in`? I simply copied the file into the same location (/canute-ui), and renamed the copy `config.rc`. Now the error generated from running `python canute_ui` is:
[image]

kasbah, 2017-08-10:
That should do it, yeah. I guess you weren't able to install pyside?

ArmanArbab, 2017-08-10:
No, on Windows, when I run `pip install pyside` I get an error saying that only certain python versions are supported. Currently I'm using python 3.6. I'll try uninstalling and using python 3.4 instead. 
[image]

ArmanArbab, 2017-08-10:
After installing python 3.4, and cmake, pip alerted me that pyside also required qmake. I then attempted to install PyQt4 from a `.whl`, but I got an error saying that it was not a supported wheel. Do you know of a binary installer for PyQt4?
[image]

kasbah, 2017-08-10:
Hmm, the only thing I could think of is Chocolatey: https://chocolatey.org/packages/pyqt4

But I have never tried it myself.

ArmanArbab, 2017-08-10:
Updating pip fixed the problem with installing PyQt4,  but I still get an error when using pip to install pyside: `Failed to find qmake`

kasbah, 2017-08-10:
Wait a minute, we are using PySide which is separate from PyQt! Qmake is part of Qt not PyQt. 

I think you can get it here: https://www.qt.io/download-open-source

ArmanArbab, 2017-08-10:
I installed qmake and added it to my system PATH variable; however, I now get a different error:
[image]

kasbah, 2017-08-10:
Oh dear, I am not sure. What does it say in the CMakeError.log that it mentions.

ArmanArbab, 2017-08-10:
The strange thing is that when I go to C:/Users/armanarbab/AppData/Local/Temp, there is no folder called `pip-build-ebvormd1`, nor any file with "CMake" in its name.

kasbah, 2017-08-10:
Sure, it would be nice for development purposes though: to have people on Windows work on the software using the emulator.

ArmanArbab, 2017-08-10:
Perhaps sometime I can repeat the above steps on CentOS and see if I can find anything meaningful in the cmake error log generated there.

pachpict, 2019-08-20:
Will not fix: Emulator not guaranteed to work on Windows, use Linux or Linux-on-Windows (in Windows 10).

closed 2019-08-20 by pachpict