Bristol-Braille/canute-ui #19: TypeError in string format in driver_pi.

kasbah, 2016-09-28:
This happened the first time I managed to press the next button with the ui running on a fresh Pi image. The ui continued to run after that. 
~~~
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 859, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 732, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 471, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 335, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Logged from file driver_pi.py, line 61
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 859, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 732, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 471, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 335, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Logged from file driver_pi.py, line 61
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 859, in emit
2016-09-28 23:44:43,183 - root             - INFO     - despatching screen->next()
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 732, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 471, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 335, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
~~~

kasbah, 2016-09-29:
This error has now stopped appearing after running the UI about 10 times.

mattvenn, 2016-09-29:
what version?
nothing doing logging on line 61.

kasbah, 2016-09-29:
I think this might have been because the time was not configured. Maybe
Raspbian ran ntp in the background which sorted it.

kasbah, 2016-09-30:
It was the latest version from git. I am probably wrong about the NTP thing as the correct time seems to be printed in what I pasted above.

kasbah, 2016-12-19:
We haven't been able to reproduce this and likely can't, as reported, with merge of #27.

closed 2016-12-19 by kasbah