Bristol-Braille/canute-ui #20: A study in Scarlet causes crash

kasbah, 2016-09-29:
I believe I added a hack to remove books that cause the UI to crash, I think A Study in Scarlet might be one such book but it still appears in the list so crashes because it doesn't exist. 

Logged from file driver_pi.py, line 61
2016-09-29 00:15:04,266 - root             - INFO     - despatching ui->load_book(6)
2016-09-29 00:15:04,269 - pageable         - DEBUG    - loading book 15 [uc A Study in Scarlet]
2016-09-29 00:15:04,274 - driver_pi        - ERROR    - OSError : [Errno 2] No such file or directory: '/home/pi/books/uc A Study in Scarlet.canute'
2016-09-29 00:15:04,277 - driver_pi        - ERROR    - closing serial port

kasbah, 2016-09-29:
I think it crashes on conversion and is thus removed. The old library refresh was used to take care of the rest but that doesn't exist anymore. We should probably find out why the conversion crashes.

kasbah, 2016-09-29:
Indeed: [pageable.py#L433](https://github.com/Bristol-Braille/canute-ui/blob/master/ui/pageable.py#L433).

Looks like the XML parser crashes on it so it's not really our fault.

kasbah, 2016-09-29:
```
ExpatError: not well-formed (invalid token): line 1, column 0
```

Can't spot any immediate XML problem with the file. In any case we need to figure out a better strategy for handling bad conversions other than just deleting the file :)

kasbah, 2017-01-04:
This doesn't crash any more but the user should be notified of bad conversion.

pachpict, 2017-01-18:
Best approach would be an error noise for each bad conversion. Then in the Library Menu books that didn't convert would also be listed. E.g.;

book 1
book 2
conversion error: book 3
conversion error: book 4

kasbah, 2017-08-08:
Bad conversions will be noticed immediately when #37 is implemented.

closed 2017-08-08 by kasbah