Bristol-Braille/canute-ui #21: UI on desktop not comunicating with physical canute.

o0Mouse0o, 2016-09-29:
The UI is not communicating with the Canute. 
It appears to be complaining of received serial string length problem. 

> 2016-09-29 17:05:02,410 - buttons_config   - INFO     - config OK
2016-09-29 17:05:02,410 - root             - INFO     - running with both emulated and real hardware on port /dev/ttyACM0
2016-09-29 17:05:02,417 - driver_pi        - WARNING  - Could not import evdev
2016-09-29 17:05:02,417 - driver           - INFO     - device ready with 40 x 9 characters
INFO:__main__:display GUI
2016-09-29 17:05:02,919 - driver_emulated  - INFO     - started qt_display.py with process id 3132
2016-09-29 17:05:02,920 - driver_pi        - INFO     - hardware detected on port /dev/ttyACM0
2016-09-29 17:05:02,922 - driver           - INFO     - device ready with 40 x 9 characters
2016-09-29 17:05:02,924 - pageable         - INFO     - library has 23 books available
2016-09-29 17:05:02,955 - root             - INFO     - starting in book mode
2016-09-29 17:05:02,956 - pageable         - INFO     - book [about ncbi word g1] open at 0 of 34 pages
2016-09-29 17:05:02,957 - pageable         - INFO     - showing page 1 of 34 (lines 0 -> 9)
2016-09-29 17:06:03,002 - driver_pi        - WARNING  - unexpected rx data length 0
2016-09-29 17:06:03,003 - root             - ERROR    - unpack requires a string argument of length 2
Traceback (most recent call last):
  File "ui.py", line 297, in <module>
    ui = UI(driver, config)
  File "ui.py", line 42, in __init__
    self.show()
  File "ui.py", line 182, in show
    self.driver.set_braille(data)
  File "/home/mouse/Documents/braille/canute/canute-ui/ui/driver_both.py", line 33, in method
    pi_method(*args, **kwargs)
  File "/home/mouse/Documents/braille/canute/canute-ui/ui/driver.py", line 132, in set_braille
    self.set_braille_row(row, row_braille)
  File "/home/mouse/Documents/braille/canute/canute-ui/ui/driver.py", line 145, in set_braille_row
    self.status = self.get_data(CMD_SEND_LINE)
  File "/home/mouse/Documents/braille/canute/canute-ui/ui/driver_pi.py", line 170, in get_data
    data = struct.unpack('2b', message)
error: unpack requires a string argument of length 2

kasbah, 2016-09-29:
Russ, could you try with the dev-kit?

kasbah, 2016-10-07:
Going to close this as this seems to have been a firmware issue that was resolved.

closed 2016-10-07 by kasbah