Bristol-Braille/canute-ui #22: Replacing all books can let you end up on non existent library page

kasbah, 2016-10-01:
If you are on page 3 of the library say, you will end up on page 3 of 1 if you only have on page in the new library. 

```
2016-10-01 18:45:17,920 - pageable         - INFO     - removing old brf file
2016-10-01 18:45:17,933 - pageable         - INFO     - adding new book [iceb_2016_designing_canute_with_the_blind_community] to library
2016-10-01 18:45:17,951 - pageable         - INFO     - showing page 3 of 1 (lines 18 -> 1)
```

kasbah, 2016-10-07:
Not  exactly sure how it happened, we have been trying to find the issue with the buttons not responding,  but on MK10 I found the UI in a crashed state with: 
~~~
2016-10-01 21:37:12,757 - root             - INFO     - despatching ui->load_book(4)
2016-10-01 21:37:13,309 - driver_pi        - ERROR    - OSError : [Errno 2] No such file or directory: '/home/pi/books/randomsc3.canute'
2016-10-01 21:37:13,321 - driver_pi        - ERROR    - closing serial port
~~~

pachpict, 2016-10-07:
Library button should always return to page one of library (and in future the same with other menus), until feedback is received to the contrary. I don't think people expect menus/directories to remember their place in it from last time.

commit d7c074fe (2016-10-10): fixes #22
closed 2016-10-10 by mattvenn