Bristol-Braille/canute-ui #24: Confirmation menu/s, waiting screens and distinguishing selectable lines

pachpict, 2016-10-07:
After selecting "Replace library" users should be taken to a confirmation menu (called "Confirm action", see issue #13) with the following;

(Non-selecting line 1) This action will remove all existing 
(Non-selecting line 2) books from the Canute's library and 
(Non-selecting line 3) add books on the USB stick.
(Selectable line 4) Yes, replace library
(Selectable line 5) No, don't replace the library

If the user selects "Yes, replace library" it can take ten minutes or more to complete the action, depending on the size of the files being loaded. So have a waiting screen along the lines of;

"Replacing the library. This can take 
several minutes per added book."

With introducing non-selectable items to menus there needs to be a common way to distinguish them. Suggest we use an asterisk (DOT-5, DOT-35; AKA ⠐⠔) for selectable items across all menus.

kasbah, 2017-08-08:
How does this fit in with #82?

pachpict, 2019-04-01:
Although the need to confirmation menus and so on remains, this issue specifically refers to now-abandoned features from old Canute prototypes, so I am closing it.

closed 2019-04-01 by pachpict