Bristol-Braille/canute-ui #34: Rolling hard reset to reduce accumilated errors

pachpict, 2016-11-29:
Whilst setting pages of text as normal, each line should do a hard reset every 9* page changes.

 - So on first page refresh first line does hard reset before setting, other lines set as normal.
 - On second page refresh second line does hard reset before setting, all other lines set as normal.
 - &c. &c.
 - On tenth page refresh first line does hard reset before setting, other lines set as normal.
 - &c. &c.

* For 9 read "number of lines", so compatible with 8-line Mk9 and future models with more or less lines.

pachpict, 2016-11-30:
Var X should be able to be modified by another var (say Y), which is a multiplier.

So Y=1 to start with, during testing, meaning X = no-of-lines * Y.

Then, if we are confident that Canute doesn't require that level of extra safety (Mk11 POC is looking very good!) we can set Y=4 (or similar) to reduce the frequency of hard re-setting.

kasbah, 2016-12-07:
Just looking over this. I believe it is better done completely in firmware as it is a low-level detail the UI shouldn't have to care about: the specifics will change depending on the hardware. 

The main problem with implementing this is that individual lines can be set. The simplest approach to this is simply rolling the refresh according to `number_of_lines_set / page_height`.  It would then go out of sync if a single line is set on it's own. Would this be an issue?

Another approach might be to count the number of times an individual line has been set and refresh that line after it reaches a threshold.

pachpict, 2016-12-15:
Yes, I was thinking second approach as well. Offset would be its line number. Threshold would be int set in config file ideally, so can be tuned on fly during testing.

This is quite an urgent task now as we have had feedback from Mk10 from New College Worcester school. They said that, unlike the Mk9, which didn't accumulate errors after initial hard reset errors, the Mk10 gets worse and worse the longer its read. After somewhere between 10--30 pages it becomes unreadable.

Therefore we should probably start with threshold of 9, then we can double it if feeling confident.

kasbah, 2016-12-15:
Ok, I will come in tomorrow and work on this.

pachpict, 2016-12-15:
Thanks Kaspar

| Bristol Braille Technology CIC | Tel: +447908 569 214 |

On Thu, 15 Dec 2016, Kaspar Emanuel wrote:

> Ok, I will come in tomorrow and work on this.

kasbah, 2016-12-16:
This has been implemented in firmware.

pachpict, 2017-01-18:
We should leave this in as an option in firmware, but not actively use it for Mk10 or Mk11s. Will favour manual hard reset by user instead.

kasbah, 2017-08-08:
This is no longer releavant for mk12.

closed 2017-08-08 by kasbah