Bristol-Braille/canute-ui #37: Read PEF and BRF files directly?

kasbah, 2016-12-07:
It seems fairly straight forward to skip the conversion step completely and simply read and convert BRF and PEF files on the fly. Is there much of a reason for the native file format other than the initial development bootstrapping and it's legacy? What other formats are on the horizon where the native files might be useful again?

pachpict, 2016-12-15:
Only other formats would be plain text files, converted directly into Grade 1 1-to-1 computer Braille.

Not sure why we had the native format. I think it may be because PEF doesn't require line breaks, and if line breaks not used is slow to parse.

kasbah, 2016-12-15:
>if line breaks not used is slow to parse.

Well, we would only every read max 40x9 characters at a time to display an entire page so I don't think it will be an issue. 

We currently have to pass the display width and height around everywhere because it's needed for the conversion and reading of the converted files but it doesn't need to be. It could simply request the number of characters it needs from a file.

pachpict, 2016-12-15:
But PEF needs to be parsed to read xml nodes first. You can't just take x number 
of characters as they may be over two lines if ignoring xml nodes.

Having said that, I've not had to manipulate xml for almost two years, so 
probably forgotten how it works.

For now the tast isn't urgent. But I agree the BRF at least could be read 
directly. Maybe BRF could be the internal format and PEF could be converted into 
it. One for a rainy day, perhaps.

| Bristol Braille Technology CIC | Tel: +447908 569 214 |

kasbah, 2016-12-15:
Hmm it may require a streaming XML parser, thought it's worth investigating how long the XML parsing actually takes currently. It's something people have spent a lot of time optimizing. 

The main driver for this issue is the time it takes to do the conversion (3 or 4 minutes currently with 20 books on the Pi?) which would be completely eliminated.

mattvenn, 2016-12-15:
it was to make it faster to load books, it took too long to load a book
after pressing the button. So yes a streaming xml parser could work.

kasbah, 2016-12-15:
Yes, but weren't we trying to convert the whole book at once back then rather than just parsing the XML and then loading and converting lines as we need them? I.e. isn't the bottle neck our conversion rather than the XML parser?

mattvenn, 2016-12-15:
dang I can't remember.

pachpict, 2016-12-15:
For what its worth, I vaguely recall us starting out reading PEF directly, then 
moving to a native format --- i.e. a cache --- due to the speed.

kasbah, 2017-01-13:
I tested with the biggest XML file we have ('uc Dracula.pef' - 3.3 mega bytes) and it took a second or two to parse the whole file with `xml.dom.mindom.parse`.

pachpict, 2017-01-16:
Then lets make the change as soon as it is appropriate and convenient to do so.

On Fri, 13 Jan 2017, Kaspar Emanuel wrote:

> I tested with the biggest XML file we have ('uc Dracula.pef' - 3.3 mega bytes) 
> and it took a second or two to parse the whole file with 
> `xml.dom.mindom.parse`.

kasbah, 2018-06-15:
This was implemented with #120.

closed 2018-06-15 by kasbah