Bristol-Braille/canute-ui #41: Random line refresh delay in latest firmware

pachpict, 2017-01-08:
Testing Mk10: On most page refreshes there will be a random delay between two lines. Cannot observe pattern. Delay is roughly 1--2 seconds. I.e. 
 - Refresh line 1
 - "" 2
 - "" 3
 - Pause for a second
 - Refresh line 4
 - &c.
Never previously observed. Assume is UI rather than firmware.

pachpict, 2017-01-08:
Bug causes serious user experience problems. For example; pausing for three seconds between rendering line two and line three in library menu gives the impression that no more books are loading, prompting user to make selection.
High priority to fix.

kasbah, 2017-08-08:
I believe these issues are caused by faulty SD cards or filesystems.

closed 2017-08-08 by kasbah