Bristol-Braille/canute-ui #42: Flakey Replace Library option, variety of symtoms

pachpict, 2017-01-08:
During testing appeared to demonstrate the following problems (some may be related or random as based off small sample size);

 - Will not Replace Library more than once per power-up.
 - Will not load PEF file with BRF files (four files, three BRF, two PEF, PEF wasn't added with repeated tests, was loaded when on its own).
 - Will not Replace Library if Library is currently empty.

kasbah, 2017-01-08:
Regarding second point, https://github.com/Bristol-Braille/canute-ui/blob/master/ui/utility.py#L33 should not be there.

kasbah, 2017-01-08:
Actually, this doesn't make much sense. Could you attach the files you were testing with please.

kasbah, 2017-01-08:
I just test scenario 3 on my laptop and it works fine. How were you testing this exactly?

kasbah, 2017-01-08:
Is it maybe that the library directory hasn't been created?

pachpict, 2017-01-11:
Kaspar

Circumstances of test were immediately prior to taking for testing, so not 
conducted with rigour or tested.

We tested the Mk10 with the UI as Russ and I found it after you were working on 
it Friday night, Saturday morning. Used all sorts of variations of the four 
files that were in the test_books dir.

Side note: That rogue commit changing the test_books dir was me. I have changed 
OS and not configured Git properly yet. Was under the impression test_books dir 
wasn't used by the system, just for storing useful test files.

BRF books were from the ICEB 2016 directory. Only changes were to remove triple 
blank lines and shorten lines with page numbers on (commonly ending "  #ab" or 
similar).

The one PEF file was generated by Dave W and has been used in Mk8, Mk9 and Mk10 
previously. Only change was to pad out one page slightly.

Initially couldn't get to work with any combination of files. We took the SD 
card out of the Pi and manually replaced library with ones form the repo. This 
then worked with "Replace Library".

However if had both PEF and BRF files on it would only load the BRF files.

If had just PEF file on it loaded that fine.

If tried to Replace Library a second time without shutting down and restarting 
and it wouldn't do it. On restart would do it.

In the end settled for the three BRF books and gave up on the PEF for the demo.

As I said, all the above was done in a rush to get a demo ready.

Ed

| Bristol Braille Technology CIC | Tel: +447908 569 214 |

kasbah, 2017-01-11:
>Initially couldn't get to work with any combination of files. We took the SD card out of the Pi and manually replaced library with ones form the repo. This then worked with "Replace Library". 

>However if had both PEF and BRF files on it would only load the BRF files.

So we should be able to reproduce this with just the files from the repo?

kasbah, 2017-08-08:
I don't think we have ever been able to reliably reproduce this issue?

closed 2017-10-19 by kasbah