Bristol-Braille/canute-ui #50: PySide missing from requirements.txt

rbubley, 2017-01-12:
PySide is imported in qt_display.py, but is not in requirements.txt

kasbah, 2017-01-12:
Thanks rbubley, it's not included to avoid long re-compilation times. It's best installed system wide with a binary release as detailed in [the installation instructions](https://github.com/Bristol-Braille/canute-ui/blob/master/INSTALL.md#install).

Additionally we don't want to install it on actual Canute machines that never need to run the emulator though we could keep a separate requirements.txt for production I suppose.

Do you know of a way to declare it in requirements.txt but avoid re-installations on systems where it already installed through apt or some other mechanism?

rbubley, 2017-01-12:
I see. I wasn't installing under linux so somehow missed that line. 

It's not uncommon to have a separate "requirements-dev.txt" to separate out dev only dependencies. 

To have more flexibilty, e.g. to make tests for system-wide versions, you would generally use setup.py rather than requirements.

kasbah, 2017-01-12:
Yeah, fair enough, if you could add installation instructions for your platform that would be very much appreciated.

closed 2019-08-20 by pachpict