Bristol-Braille/canute-ui #55: Displaying graphics

rbubley, 2017-01-12:
Any interest in being able to display graphics? I could put together a PR.
[image]

kasbah, 2017-01-12:
That's really cool! Could you describe more what the input could be? 

I think it could be a good feature to demo for sure and then gather some feedback. @pachpict might already have some insight into what the ability to add graphics like this could be used for.

rbubley, 2017-01-12:
Input can be pretty much anything you like: it uses Pillow (
http://pillow.readthedocs.io/), which has plugins for practically every
graphics format out there (most of which will be compilable for a raspberry
pi).

When I get a little time I'll clean the code up and submit a PR.

rbubley, 2017-01-13:
PR #56

pachpict, 2017-01-16:
The potential for certain types of graphics to Canute is very exciting, and entirely unexplored by the BBT team so far. If we could come up with a variety of test files (SVG, PNG &c) for different types of data (photo, diagram, logo &c) then it would be a hugely interesting project for the Braillists to pursue.

pachpict, 2017-01-16:
Hi Andrew

Thanks for getting in touch. The Canute has the following properties (seen from
the point of view of tactile graphics, rather than Braille);

 - Horizontal: 80 dots across, spaced 2.5mm, 4mm, 2.5mm, 4mm &c
 - Vertical: 54 dots down, spaced 2.5mm, 2.5mm, 15.5mm, 2.5mm, 2.5mm, 15.5mm &c
  
It is therefore possible to display tactile graphics on the display, but a
straight-forward conversion (either stretching the image over the 15.5mm spaces
or losing data there) would be unlikely to work for anything more than simple
shapes. Therefore it is likely to require more abstracted take on tactile
graphics.

In the example of maps it may, for example, be better suited to showing the
London Metropolitan Line tube map, including station names, than it would be a
geographically accurate road map.
 
There is a discussion of this subject on our Open Sourced code repository here:
https://github.com/Bristol-Braille/canute-ui/issues/55
 
As well as responding to you I will add my response to your email to the thread
there.  
 
Sincerely

Ed

> Hi Canute developers
>
> Do you think the Canute has the potential to display tactile maps?
> 
> Kind regards
>
> andrew

closed 2019-04-01 by pachpict