Bristol-Braille/canute-ui #57: Using sockets for emulated display is problematic

rbubley, 2017-01-13:
The process for communicating with the emulator display using sockets is flaky: fundamentally, sockets offer no acknowledgement nor retransmission opportunities, so messages between the emulator GUI and the UI process can be missed, or may be delayed pending buffers being filled/flushed. 

For example if you create a GUI and try to display something immediately, it may not get displayed because of a race condition in getting the sockets set up versus a timeout.

If a higher-level messaging protocol is used, for example using Python's standard multiprocessing module, then these problems would go away.

kasbah, 2017-01-13:
This is true rbubley. I think a PR for better emulator communication would be accepted. 

The current scheme does work quite well for our purposes so far and it has way less limitations than the communication with the actual machine. 

With that in mind I think for work in this direction to be worthwhile it should keep or reduce the complexity of the application. I.e. I don't want to trade added complexity for robustness in this instance.

kasbah, 2017-05-31:
This is solved with #58

closed 2017-05-31 by kasbah