Bristol-Braille/canute-ui #77: Canute UI MK12 Bookmark proposal

cz1392, 2017-06-27:
[image]

kasbah, 2017-06-27:
Sorry, but why is this a picture of text?

pachpict, 2017-06-27:
Something weird to do with copying and pasting from MS Word on Mac OS into the Github site. Very odd. I gather Chen'll have another go tomorrow copying from a text editor instead.

cz1392, 2017-06-28:
Hi, I think the current output should be text.

kasbah, 2017-07-17:
Pressing the home button goes back to the book, right?

pachpict, 2017-07-20:
> Pressing the home button goes back to the book, right?

Yes, always. If already in the current book then it toggles the menu 
option labels.

kasbah, 2017-08-07:
>Feedback present as changing the whole content to bookmark added then back to the current page of the book.

What do you mean by this @cz1392?

steph-sergeant, 2017-08-07:
Hi Kaspar, 

Chen suggested this. 
I take it to mean, that the display will say 
'bookmark added'  
Then return to the book. 

Another possible solution would be to change just the first few cells on the first line to 
'bookmark added' 
For say half a minute, then return to the book. 

I still cannot read your comments on the Github,  just the emails that come. 
 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-07:
There is nothing about the bookmark menu title, should it just be a list of bookmarks or should we have a title that says "Choose a bookmark" or something similar? Should the book title be shown somewhere in the bookmark menu?

kasbah, 2017-08-07:
If we don't have a title then how do we display the bookmark menu page number?

steph-sergeant, 2017-08-07:
I thought we have the title of the book on the first line. 

Chen, I still cannot read your messages, Kaspar's are appearing in my inbox, but not yours. 

I cannot read the issues on the internet. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-07:
Steph, you can only read new messages since you have subscribed. Here is what Chen wrote:

> Bookmark proposal
> 
> The bookmark is for a page.
> 
> N.B. Bookmarks apply to a whole Canute page (not line by line or BRF page)
> 
> - [ ] - Insert a bookmark:
> 
> Press button 5 in home menu to set a bookmark.
> 
> Feedback present as changing the whole content to bookmark added then back to the current page of the book. 
> 
> - [ ] - Find a bookmark:
> 
> In home menu press button 6 to go to bookmark menu. 
> 
> In bookmark menu, each bookmark will present in one line. Canute present the first 10 cells about bookmark location, the rest cells show the first line of the content for this page. 
> 
> Press side bat button to go to the certain page.
> 
> N.B. No limit to the number of bookmarks per book. The user can scroll around bookmarks with left and right buttons from front buttons.
> 
> - [ ] - Delete a bookmark:
> 
> In bookmark menu, long press side bar buttons for certain bookmark to delete that bookmark.
> 
> N.B. Sorting can be changed in system menu between by page number and by last visited.

steph-sergeant, 2017-08-07:
Thanks Kaspar, very helpful indeed. 

I can read the Seafile documents with ease by the way.

Can you email me anything else that Chen has put up in the last month please?

Thanks 
Steph 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

cz1392, 2017-08-08:
When we designed the bookmark, we assumed the bookmark is for the current book. So for the title, I think the first line can display as 'Chose the bookmark from sidebar button'.

For each existed bookmark, I think it displays the location in first 10 cells for that line, 

e.g. 90/1000
90 is the bookmark page number
1000 is the total page number

then the rest cells for that line show the content for that page with where the content first appeared on that page.

cz1392, 2017-08-08:
The feedback purpose is how to let the reader know if the bookmark has been added properly.

kasbah, 2017-08-08:
This has been implemented with #96.

For the title I put : ",,bookmarks: <book title> page / total", if there is only one page no page / total numbers are displayed. 

For the bookmarks I have put page number and then excerpt from the first line of that page. I think the repetition of the total page numbers on every bookmarks e.g. 90/1000 then 98/1000 etc. would be wasting a lot of space.

closed 2017-08-08 by kasbah