Bristol-Braille/canute-ui #78: Canute UI MK12 Home menu proposal

cz1392, 2017-06-28:
Home menu

Long press to go 5 pages forward or backwards at front buttons. 

Access to Home menu
Press menu button (central button from the front buttons)then press sidebar.
Press menu button again, it goes back to the current book.

Line 1 displays’ name of the current book and current location’, button 1 will do nothing.
Line 2 displays ‘Go to page’. A press of button 2 will achieve this function. Specifications are shown in ‘Go to page menu proposal’.
Line 3 displays ‘Go to start of the book’. A press of button 3 will achieve this function.
Line 4 displays ‘Go to the end of the book’. A press of button 4 will achieve this function.

Line 5 displays ‘Insert Bookmark for current page’. A press of button 5 will achieve this function.
Line 6 displays ‘Go to Bookmark’. A press of button 6 will achieve this function.

Line 8 displays ’Go to library menu’. A press of button 8 will achieve this function.
Line 9 displays ’Go to system menu’. A press of button 9 will achieve this function.

kasbah, 2017-07-17:
So line 7 is just blank and no binding?

kasbah, 2017-07-17:
Why would someone want to go to the end of a book?

steph-sergeant, 2017-08-07:
Because they are looking for information that they know is near the end of the book. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-07:
But surely that's quite a rare case and doesn't warrant it's own menu option. They can easily go to a specific page number, including the last page, using the go to page system.

steph-sergeant, 2017-08-07:
Yes, that is true, 
We did put that in, when we were not sure if you could do page numbers.
Now you can, perhaps we can leave it out. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-07:
Hi Kaspar, 

I have been talking to another Braille reader, and End of book would be useful to her too. 

This is quite a common command in the other Braille note takers, and is far quicker and easier to use than the go to page. 
I would like to add it to chen's questionnaire please. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-07:
Ok, I will implement it. It's not very hard to add, it just struck me as odd.

steph-sergeant, 2017-08-07:
You are a star. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-08:
I changed some of the wording slightly, hope that's ok:

1. go to page
1. go to start of book
1. go to end of book
1. insert bookmark at current page
1. choose from existing bookmarks
1. 
1. view library menu
1. view system menu

Otherwise this is all done with merging of #96

closed 2017-08-08 by kasbah

kasbah, 2017-08-08:
>I agree mostly with what you have done.
>The fact that you have changed the numbering, means I will need to change several help menus.

Sorry, the numbering actually hasn't changed. We still have the book title with page number as the first line and the rest are offset, as before. I only changed the wording a bit. 
>
>How about we put the literal
>'home menu' centralised on the top line, then we can move everything down one number, and I will not need to change the sequencing.
>I agree it is meaningless to have a title of a book on the top line, as this is a Home menu, like a central one.
>To have its name makes it so much easier to identify when directing people by phone.

I think we should leave the book title with page number as it is.  It's the only place to easily see which book you are reading and what page you are on. We do have one spare line so we could put HOME MENU above that.

steph-sergeant, 2017-08-09:
> I have edited and simplified tasks as follows:
>
> 	*	Book/Home menu (done)
> 	*	Library (done)
> 	*	Go to page (not done)
> 	*	Bookmark menu (not done)
> 	*	Library menu (not done)
> 	*	System menu (full user guide -- not done)
>
> I couldn't find texts for most options except book/home and library. I have implemented those with following texts:
>
> Home/book:
>
> 	Access to Home menu is gained by:
> 	Pressing menu button (Has Letter M on it.

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

-----Original Message-----
From: Stephanie Sergeant [mailto:stephanie@screenreader.net] 
Sent: 09 August 2017 12:49
To: Kaspar Emanuel (kaspar.bumke@gmail.com)
Subject: FW: [Bristol-Braille/canute-ui] Help menu (#84)

From subsequent emails, looks like you have done some more of these menus. 

 

 

Stephanie Sergeant 

Director

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email: Stephanie@screenreader.net 

Website: http://www.visionthroughsound.co.uk

	Pressing menu button again goes back to
	the current book.

Library:

	Choose the book you wish to read by
	pressing the button to the left of the
	title. Press the right arrow button to
	find more books in the library, if there
	are more than 8. If you cannot find the
	book you are looking for, you may wish
	to change the order in which the titles
	appear, by going back to the system menu
	and Choosing a different order.

	(page 2)
	The system menu can be reached by
	pressing the button with the letter M
	twice on it, then choosing button 9
	where the line says "System Menu". The
	choices are "Most recently read, at the
	top" or "In alphabetical order"

I think it these need to be kept more clear and snappy and should be restricted to one page:

Home/book:

	Access the home menu by pressing the
	menu button. The menu button has the
	letter M on on it and is in the middle
	of the two arrow buttons at the bottom.
	Pressing the menu button again always
	returns you to your current book.

Library:

	Choose the book you wish to read by
	pressing the button to the left of the
	title. Use the arrow buttons to page
	through the library. You can change the
	ordering of the books in the system
	menu.

What do you think?

	N.B. If Help button is pressed again, it goes back to the current book. If the central button is pressed it will return to the current book.

I have made it so the Help button returns to the current menu, and the menu button, as always returns to the book. Otherwise you go to get help on the library menu for instance, but have no quick way to go back to the library menu which is not a good user experience.

steph-sergeant, 2017-08-09:
Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

-----Original Message-----
From: Stephanie Sergeant [mailto:stephanie@screenreader.net] 
Sent: 09 August 2017 12:48
To: Kaspar Emanuel (kaspar.bumke@gmail.com)
Subject: FW: [Bristol-Braille/canute-ui] Help menu (#84)

Where are you picking up the text from? 

 

I have written help text in Seafiles ui design, documents starting with ‘help’.  

 

The menu button now has the word menu on it, rather than just M. 

 

Yes, possibly your snappier text is better. Sometimes though when people are stuck, they need things spelt out. 

 

 

Stephanie Sergeant 

Director

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email: Stephanie@screenreader.net 

Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

-----Original Message-----
From: Stephanie Sergeant [mailto:stephanie@screenreader.net] 
Sent: 09 August 2017 12:38
To: Kaspar Emanuel (kaspar.bumke@gmail.com)
Subject: FW: [Bristol-Braille/canute-ui] Home menu (#78)

Ok, I will update the home menu in Seafile u.i. design. 

 

 

Stephanie Sergeant 

Director

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email: Stephanie@screenreader.net 

Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
> 	Canute to come with user guide; hardcopy and BRF copy on the Canute. BRF copy on the Canute to be pre-loaded internal to the Canute, not on SD card like other user files. First thing user sees when opens up Canute.
>
> 	Some suggestions about the contents:
>
> 	What should it contain?
>
> 	Getting start
>
> 	How to access to PC?
>
> 	How to insert book to Canute?
>
> 	What file format does Canute support?
>
> 	What’s the main function for each button?
>
> 	Customer service
>
> Does this need to be tracked on GitHub? It doesn't seem like a coding task.

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

-----Original Message-----
From: Stephanie Sergeant [mailto:stephanie@screenreader.net] 
Sent: 09 August 2017 12:36
To: Kaspar Emanuel (kaspar.bumke@gmail.com)
Subject: FW: [Bristol-Braille/canute-ui] User Guide (#83)

Most of this is already written, I need to finish it and convert to BRF. 

 

 

Stephanie Sergeant 

Director

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email: Stephanie@screenreader.net 

Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

-----Original Message-----
From: Stephanie Sergeant [mailto:stephanie@screenreader.net] 
Sent: 09 August 2017 12:33
To: Kaspar Emanuel (kaspar.bumke@gmail.com)
Subject: FW: [Bristol-Braille/canute-ui] Confirmation menu/s, waiting screens and distinguishing selectable lines (#24)

If you give me more detail, I will try and enlighten you.

 

 

Stephanie Sergeant 

Director

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email: Stephanie@screenreader.net 

Website: http://www.visionthroughsound.co.uk

 

How does this fit in with #82 <https://github.com/Bristol-Braille/canute-ui/issues/82> ?

steph-sergeant, 2017-08-09:
> With the system menu being cleaned up I think we should have a developer menu activated through some special button combination unlikely to be pressed by the user by accident. Items would currently be:
>
> 1.	Update UI
> 2.	Copy log to USB stick
> 3.	Reset display
> 4.	Display IP address

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

-----Original Message-----
From: Stephanie Sergeant [mailto:stephanie@screenreader.net] 
Sent: 09 August 2017 12:30
To: Kaspar Emanuel (kaspar.bumke@gmail.com)
Subject: FW: [Bristol-Braille/canute-ui] Developer menu (#101)

Brilliant idea, go for it. 

 

I agree no point in having the serial number at the beginning of the book, title more important.

 

Are you implementing help text? Looks like it from the emails. 

If so, I do not need to print out 10 copies of all the help menus. 

 

Please let me know if they will be displayed on the MK 12 for Thursday. 

 

Thanks Steph 

  

 

Stephanie Sergeant 

Director

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email: Stephanie@screenreader.net 

Website: http://www.visionthroughsound.co.uk