Bristol-Braille/canute-ui #79: Canute UI MK12 'Go to page' Proposal

cz1392, 2017-06-28:
Go to page menu proposal

N.B. The page number is shown as Canute page number rather than BRF page number. 

If Button 2 in home menu is pressed, goes to ‘Go to page menu’
Line 1 displays ‘Enter page number from the sidebar to go to the desired page number’. 
Line 2 displays current location with total page number
Line 3 displays the feedback from the side bar. (Format like Go to page number __ /total)
Line 4 shows please confirm by pressing the forward button from front buttons. 
Line 5 shows please enter a different page number by pressing backwards button from front buttons.
Line 6 shows go back to the current page in a book by pressing central button from front buttons.
Insert page number by side bar buttons.

Page numbering is up to page 9999. If the user enters eg 14 then this is treated the same as entering 014, i.e page 14. N.B. at 

PS. Likely problem: May confuse readers. 
Possible solution: put BRF page number as an option in system menu

kasbah, 2017-07-17:
>‘Enter page number from the sidebar to go to the desired page number
Is 67 characters and won't fit on one line with direct transcription.

kasbah, 2017-07-28:
This has been implemented with #93.

closed 2017-07-28 by kasbah