Bristol-Braille/canute-ui #80: Canute UI MK12 Library menu proposal

cz1392, 2017-06-28:
Library proposal

Library only shows the book in selected sorting method from the system menu.
Titles of the book finishes with current page /total pages (1 line)

Otherwise same as current implementation

cz1392, 2017-08-08:
With current feedback, serial numbers will be presented at the beginning, followed by the title of the book. If the length of the title is over the capacity of 1 line. Full title will be displayed until it finished. current page/total page number is also displayed. 

The side bar will present the among the library can 
e.g.

1. Machine learning volume 1 90/200 
2. Machine learning volume 2 70/230
3. Pattern recognition and machine learning 30/210
4. The Elements of Statistical Learning:
     Data Mining, Inference, and Prediction. 40/200

kasbah, 2017-08-08:
I am closing this in favor of more focused issues #98 and #99.

closed 2017-08-08 by kasbah