Bristol-Braille/canute-ui #83: Canute UI 'User Guide' proposals

cz1392, 2017-06-28:
Canute to come with user guide; hardcopy and BRF copy on the Canute. BRF copy on the Canute to be pre-loaded internal to the Canute, not on SD card like other user files. First thing user sees when opens up Canute.

Some suggestions about the contents:

What should it contain?

Getting start

How to access to PC?

How to insert book to Canute?

What file format does Canute support?

What’s the main function for each button?

Customer service

kasbah, 2017-08-08:
> Canute to come with user guide; hardcopy and BRF copy on the Canute. BRF copy on the Canute to be pre-loaded internal to the Canute, not on SD card like other user files. First thing user sees when [...]

Does this need to be tracked on GitHub? It doesn't seem like a coding task.

pachpict, 2019-08-20:
Manual is still a work in progress...
Important it remains in the Github issues list because many changes, quirks and features that are only recorded here need to be written into the manual.

The current version of the manual is: http://bristolbraille.co.uk/support/

zwbrbr, 2019-09-25:
Changes moving to being tracked via e-mail.

closed 2019-09-25 by zwbrbr