Bristol-Braille/canute-ui #84: Canute UI MK12 'Help menu' proposal

cz1392, 2017-06-28:
5 contextual help articles are provided in Canute. Within each help, article user can go back or forwards through pages, though articles should try and be one page. When revisiting a contextual help article the Canute does not remember where the user was before. Press 'Help button' to go to each article.

In a book or home menu, if Help button is pressed, basic navigation is presented. How to go to different menus, what is in the navigation menu, what is in library menu and what is in system menu are presented. 

In Go to page menu, if Help button is pressed, details about Go to page menu are presented. 
Basically, how to use the sidebar and how to go to the certain page are presented. 

In Bookmark menu, if Help button is pressed, how to delete a bookmark, how to add a bookmark, details about how to change bookmark order and where to change it are shown. 

In Library menu, if Help button is pressed, how to access to a book in the library and details about how to change book order and where to change it are shown. 

In System menu, if Help button is pressed, a full user guide will be presented.

N.B. If Help button is pressed again, it goes back to the current book. If the central button is pressed it will return to the current book.

cz1392, 2017-07-28:
Library menu help

Choose the book you wish to read by pressing the button to the left of the title. 
Press the right arrow button to find more books in the library, if there are more than 8. 
If you cannot find the book you are looking for, you may wish to change the order in which the titles appear, by going back to the system menu and Choosing a different order.
The system menu can be reached by pressing the button with the letter M twice on it, then choosing button 9 where the line says ‘System Menu’.  The choices are ‘Most recently read, at the top’ or ‘In alphabetical order'.

cz1392, 2017-07-28:
Home menu (book menu) help text
Access to Home menu is gained by:
Pressing menu button (Has Letter M on it. central button from the front buttons)
Pressing menu button again goes back to the current book.
Line 1 displays’ name of the current book and current location’, button 1 will do nothing.
Line 2 displays ‘Go to page’. A press of button 2 will achieve this function. 
Specifications are shown in ‘Help go to page menu’. 
Line 5 displays ‘Insert Bookmark for current page’. A press of button 5 will achieve this function.
Line 6 displays ‘Go to Bookmark’. A press of button 6 will achieve this function.
Line 8 displays ’Go to Library menu’. A press of button 8 will achieve this function.
Line 9 displays ’Go to System menu’. A press of button 9 will achieve this function.

kasbah, 2017-07-28:
@cz1392  thanks for putting these on GitHub but could you be less verbose when writing these please. Most of them are quite hard to parse and make use of while developing. You could take a look at how I reduced down #78 to the essentials as a template.

cz1392, 2017-07-28:
Thank you Kaspar
I will do in that way.

kasbah, 2017-08-08:
I have edited and simplified tasks as follows:

> - Book/Home menu (done)
> - Library (done)
> - Go to page (not done)
> - Bookmark menu (not done)
> - Library menu (not done)
> - System menu (full user guide -- not done)

I couldn't find texts for most options except book/home and library. I have implemented those with following texts:

Home/book:
>Access to Home menu is gained by:
>Pressing menu button (Has Letter M on it.
>central button from the front buttons)
>Pressing menu button again goes back to
>the current book.

Library:
>Choose the book you wish to read by
>pressing the button to the left of the
>title.  Press the right arrow button to
>find more books in the library, if there
>are more than 8.  If you cannot find the
>book you are looking for, you may wish
>to change the order in which the titles
>appear, by going back to the system menu
>and Choosing a different order.
>
>(page 2)
>The system menu can be reached by
>pressing the button with the letter M
>twice on it, then choosing button 9
>where the line says "System Menu". The
>choices are "Most recently read, at the
>top" or "In alphabetical order"

I think it these need to be kept more clear and snappy and should be restricted to one page:

Home/book:
>Access the home menu by pressing the
>menu button. The menu button has the
>letter M on on it and is in the middle
>of the two arrow buttons at the bottom.
>Pressing the menu button again always
>returns you to your current book.

Library:
>Choose the book you wish to read by
>pressing the button to the left of the
>title. Use the arrow buttons to page
>through the library. You can change the
>ordering of the books in the system
>menu.

What do you think?

> N.B. If Help button is pressed again, it goes back to the current book. If the central button is pressed it will return to the current book.

I have made it so the Help button returns to the current menu, and the menu button, as always returns to the book. Otherwise you go to get help on the library menu for instance, but have no quick way to go back to the library menu which is not a good user experience.

kasbah, 2017-08-09:
>From subsequent emails, looks like you have done some more of these menus.

Steph, please reply to the github.com address so everyone else can see your replies. I still only have a  text for two of the menus, as mentioned before. 

I looked through the Seafile documents but it wasn't clear what was supposed to go in the help text for other menus. Can you please copy in the text you think should go in for the remaining menus in a reply. Remaining menus are:

- Go to page
- Bookmark menu
- Library menu
- System menu (full user guide?)

steph-sergeant, 2017-08-09:
Bookmark help: 
Bookmark Help to appear when the Help button is pressed after button 5 or 6 have been pressed on Home menu. 

The bookmark is for page.
N.B. Bookmarks apply to a whole Canute page (not line by line or BRF page)
To insert a bookmark:
Press button 5 in home menu to set a bookmark, feedback present as changing the whole content to bookmark added then back to the current page of the book.
Find a bookmark:
In home menu press button 6 to go to bookmark menu.
In bookmark menu, each bookmark will present in one line. Canute presents the first 10 cells about bookmark location, the rest of the cells show the first line of the content for this page.
No limit to the number of bookmarks per book. The user can scroll around bookmarks with left arrow and right arrow buttons from front buttons.

Delete a bookmark:
In bookmark menu, long press side bar buttons for certain bookmark to delete that bookmark.
N.B. Sorting can be changed in system menu between by page number and by last visited.

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
Go to page menu Help 

N.B. The page number is shown as Canute page number rather than BRF page number.
This means that a page has 9 lines as shown on the Canute display. It is not possible to navigate by page number in the Braille file, or by print page number. However this type of page numbering will mean that you can  skip through sections of a book to reach the part you are interested in. 

If Button 2 in home menu is pressed, the Canute will display the ‘Go to page menu’

How to insert page number by side bar buttons.
Each of these have a Braille letter written on them, so page 59 is reached by pressing button E followed by button I following the Braille numbering system.
You may notice that 0 is shown by J on a square button, this is because  it does not correspond with any of the lines on the display.

Page numbering is up to page 9999. If the reader enters eg 14 then this is treated the same as entering 014, i.e page 14.

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
Library Menu help. 
Library only shows the book in selected sorting method from the system menu.
Titles of the book finishes with current page /total pages (1 line)

Choose the book you wish to read by pressing the button to the left of the title. 
Press the right arrow button to find more books in the library, if there are more than 8. 
If you cannot find the book you are looking for, you may wish to change the order in which the titles appear, by going back to the system menu, and Choosing a different order.
The system menu can be reached by pressing the button with the word ‘menu’ on it, then choosing button 9 where the line says ‘System Menu’.  
	The choices are  ‘most recently read, at the top’ or ‘In alphabetical order.’ 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-09:
Thanks for copying these here Steph. I had seen them on Seafile but didn't think they were supposed to be help texts because of odd phrasing throughout. 

>Bookmark help: Bookmark Help to appear when the Help button is pressed after button 5 or 6 have been pressed on Home menu. 

Button 5 would insert a bookmark and return to the book. I don't know how I would display the relevant help text after pressing it. Maybe we should add a note about how to access bookmarks to the book/home help?

>The bookmark is for page.

That's a strange statement. While technically true I don't understand what it is trying to convey. 

> N.B. Bookmarks apply to a whole Canute page (not line by line or BRF page) 

You want to start the help text with a nota bene ?

>To insert a bookmark: Press button 5 in home menu to set a bookmark, feedback present as changing the whole content to bookmark added then back to the current page of the book. 

This is an example of strange phrasing that makes me think these are not help texts but implementation notes. 

I think it would be good to try and re-write these and cut them down to succinct descriptions of what the user can do. E.g. "Add a bookmark by pressing button 5. " "Select a bookmark by pressing the button to the left of it" "Delete a bookmark by holding down the button to the left of it" etc. I still advocate for having these texts limited to one page as well.

kasbah, 2017-08-09:
The library one is OK, by the way. Well, I would still prefer my single page editing: 

> Choose the book you wish to read by
> pressing the button to the left of the
> title. Use the arrow buttons to page
> through the library. You can change the
> ordering of the books in the system
> menu.

That's 6 lines so there is still more space if you have anything to add. We could use contractions as well I suppose. Is there any issue with using contractions in the help texts?

steph-sergeant, 2017-08-09:
The following text shows system menu help followed by a description of the unit. 

System menu help .

Important note: 
Press the triangle with the letter a on it to shut down the Canute. 
There is a computer inside, and it does not take kindly to being powered off with the switch. 

Line 2 will copy all books from the SD card to the Canute library, when the triangle with letter b is pressed. 
Beware, all previous contents of the library will be overwritten. 

Line 3 will copy the contents of the Canute library to the SD card, and can be achieved, by pressing the triangle with letter ‘C’ on. 

Lines 4 and 5 give choices of how the library can be sorted. Pressing button ‘d’ or ‘e’ will sort the books, and land you in the library menu. 

Don’t worry, you need not change the sorting method, and may not need this, especially if you only have 8 books on your SD card. 

Lines 6 and 7 give choices of sorting bookmarks that you have entered in the books. Pressing buttons ‘f’ and ‘g’ will achieve  your chosen sorting method.

'Help manual description of unit

Welcome to your Canute Braille reader. 

The Canute is designed with reliance on users’ feedback and suggestions so do get in touch if you would like to see improvements, and we will see what is possible.
 
Place the Canute on the desk in front of you, with the curved edge nearest.

Description of unit: 

TOP: 

The Braille cells are laid out as a rectangle in Landscape format. There are 9 lines of 6 dot Braille cells, each line has 42 cells.  Along the left hand side of this rectangle are a column of triangular buttons.  These are numbered 1 to 9 in Braille, using the letters ‘a’ to ‘I’. Each button on the left corresponds to the Braille line to the right of it.

) Below I is a square button with J used for 0, with no corresponding line. 

On the top right corner is a circular button with the letter H on it. This is the Help button, and can be pressed at any time, when operating the Canute, and useful information will be displayed. 

Nearest to you are 3 buttons, in a row beneath the Braille display. One arrow to the left, that moves backwards through a book page by page, and one arrow to the right that moves forward page by page. 

Between these, is a square menu button, with ‘menu’ embossed on it. 

SIDE Panel: 
Left hand end is the SD card slot where you will need to insert the SD card with the library of books you would like to read. 

There is a mini USB port to connect to the computer.

Right hand side: 
is the power socket for the mains charger.  To the right of that is
the on/off switch, which is round. 
Plug the Canute into the mains, and press the ON switch. When released, the switch will not protrude as far.

The mains charger has Canute written in Braille on the transformer, and CAN on the plug.' 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
Please can we have the option to switch between the 2? 

If we have to choose, grade 1 might be preferable. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
System menu help.

Press the triangle with the letter a on it to shut down the Canute.
In the same way as you would on a normal computer.  

Press triangle with letter B to copy all books from the SD card to the Canute library, 
Note, all previous contents of the library will be overwritten. 

Press letter C to  copy the contents of the Canute library to the SD card.

Pressing button ‘d’ will sort the books, in alphabetical order, taking you to the library menu. 

Pressing button ‘E’ will sort the books, in most recently read first order, taking you to the library menu. 

Pressing button F sorts bookmarks into the order they appear in the book. 

Pressing button ‘g’ sorts bookmarks into most recently added first.'

Description of unit is still the same. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-09:
> Mobile: 07793 906 745 <07793%20906745>
> iks5sWZ69gaJpZM4OIFJh> . <https://github.com/notifications/beacon/

>Please can we have the option to switch between the 2?
>If we have to choose, grade 1 might be preferable.

Seems like avoiding it altogether would be preferable. Ed mentions it would
be difficult for total beginners as well.

steph-sergeant, 2017-08-09:
Go to page menu help  

The ‘page’ referred to, is the 9 line display that appears on the Canute. 

How to insert page number:
Each triangle is labelled with a Braille letter, from ‘a’ to ‘I’. so page 59 is reached by pressing button E followed by button I.  
J is a square button, used for the number 0, but not to navigate to a book. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

steph-sergeant, 2017-08-09:
Bookmark menu help: 

The bookmark is for page, the 9 lines displayed on the Canute. 
It is not possible to bookmark a specific line. 
Find a bookmark:
each bookmark will present in one line. Canute presents the first 10 cells about bookmark location, then the first words from the page bookmarked are displayed. 
You can scroll around bookmarks with left arrow and right arrow buttons. 
Delete a bookmark:
long press the corresponding triangle to that Bookmark. 
Sorting can be changed in system menu. See system menu and the help text.   

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-09:
What do you think about "displayed" as opposed to "present"? I.e. "Each
bookmark will be displayed on one line".

kasbah, 2017-08-09:
> Press letter C to copy the contents of the Canute library to the SD card.
> Pressing button ‘d’ will sort the books, in alphabetical order, taking you to the library menu.
> Pressing button ‘E’ will [...]

It seems quite odd to me to describe all the menu options again in the help. It doesn't seem add anything. I think we need to think about where the user first opens the help, explain how to use _any_ menu at that point and build on that in subsequent explanations.

kasbah, 2017-08-09:
>Each triangle is labelled with a Braille letter, from ‘a’ to ‘I’. so page 59 is reached by pressing button E followed by button I.

Maybe we could save ourselves a lot of explanations by simply numbering the buttons rather than putting letters on them?

steph-sergeant, 2017-08-09:
We did that to start with, then the engineers said there is only room for one letter. 

Braillists know that the first 10 letters form the numbers. 

Stephanie Sergeant 
Director

Vision Through Sound CIC
Tel: 0121 445 2161 
Mobile:  07793 906 745 
Email: Stephanie@screenreader.net 
Website: http://www.visionthroughsound.co.uk

kasbah, 2017-08-22:
These have now all been written and added.

PR #114 (2017-08-22): Update and enable help menu texts
fixes #84
closed 2017-08-22 by kasbah