Bristol-Braille/canute-ui #112: Popup "done" at the end of the line when inserting bookmarks

kasbah, 2017-08-10:
When inserting bookmarks within the home menu "done" can popup at the end of the line to provide feedback that a bookmark was inserted.

Lazarus404, 2018-07-28:
I'll take this one.  Will create an initial PR shortly

zwbrbr, 2019-05-09:
@pachpict Clarifying:
 * popping up text obviously doesn't work if a sly user presses the bookmark key directly while a book page is showing
 * assuming the intention was to have `INSERT BOOKMARK AT CURRENT PAGE` change into `INSERT BOOKMARK AT CURRENT PAGE ... DONE`: there's not a huge amount of space at the end of that line in Grade 1 mode (just enough, really); and in other languages there may not be enough

So I could add this as a display change, but perhaps it should (become dependent upon and) use the vibration motor instead to be more consistent?  We'd have to bump up the priority of vibration motor support, of course, so it would take longer.

zwbrbr, 2019-09-17:
@pachpict Bump.  Need to decide on the detail.