Bristol-Braille/canute-ui #128: Does BRF handling treat lower case letters as interchangable w/ capitals? (Which would be correct behaviour.)

pachpict, 2017-12-05:


pachpict, 2018-08-16:
Relates to question #165

zwbrbr, 2019-03-21:
* **Status quo**: alphabeticals are treated identically.  Before firmware release v1.2.0 only the alphabeticals were, and they'd simply be uppercased before being mapped.  (Out of scope here, but from firmware release v1.2.0 as sent out to CSUN and APH we map 0x60 onwards back to 0x40 onwards, which achieves the same thing but includes `{|}~` too, per #178)

* **What's correct**: Re-quoting James Bowden from #165: "The NABCC defines 0x60-0x7e as exactly the same as 0x40-0x5e, but without the extra dot 7."  That would mean it is correct for them to be identical on our six-dot display.

closed 2019-04-01 by pachpict