Bristol-Braille/canute-ui #132: Handle files with same filenames on separate drives

kasbah, 2018-01-09:


zwbrbr, 2019-05-09:
@pachpict This probably needs detail on what "handle" means; what would you feel is intuitive (or at least defensible) when there are dupes?  I think from #183 we roughly decided we wanted files on SD to trump (and hide) files on front-USB, which in turn trump files on back-USB (this is currently how priority works for current book and for current language).

Current behaviour, to be clear, is that:
 * books from all media are combined into one list
 * library menu entries do not reveal the medium; we have discussed a prefix to show the medium, but space on a line is already at a premium
 * so two books with the same name will simply appear to be duplicate entries
 * I'd have to check whether we're sorting the list by basename or by whole path; if by basename then duplicates will be adjacent, otherwise files will implicitly be grouped by medium

pachpict, 2022-02-11:
Should become redundant when #250 is implemented, but need to confirm that as background look loading may still use flat file structure.