Bristol-Braille/canute-ui #134: Allow user to 'Go to print page number' within book

pachpict, 2018-01-09:
Braille files often have the corresponding print page number (from the source file) on each page. This helps users navigate to the correct content when content is being referred to by its source page number (very common in classrooms).

BRFs do not support metadata, and PEFs do not usually implement this as metadata. So in Braille files you can only navigate to print page number by assuming a commonly used string that prefixes the number.

The interface for the user to be the same as conventional 'Go to page number' menu, but in the back end it would be performing a string search.

There is some confusion as to which string to search for.

George Bell of Technovision says in Duxbury the string is LEA#[NUM] (⠇⠑⠁⠼[NUM]).

Mike Townsend from Torch says the standard is dots 4, 5, 6, 2[NUM] (⠸⠂[NUM]).

I cannot find examples of either in our Canute books we've collected.

We can start by implementing the latter and check correct use once that is done.

pachpict, 2018-05-01:
Another supposed way of representing it is PP#[NUM]. 

I've asked about common strings on the Braillists Forum (https://groups.google.com/forum/#!topic/braillists/hVPcLFBbCqE).

Once we have the most common strings can implement the above feature using the most common strings, as agreed by the Braillists Forum members, which are encoded into the UI on the Pi. (NB, searches for any of the strings, even though wouldn't expect more than one style in any one book.)

But as many print houses / schools will use different styles we will need to implement another feature in the user config file on the SD-card, which would over-ride the default strings;

print-page-strings=[" LEA#", " PP#", " FAKEEXAMPLE#", etc]

Something along those lines anyway.

Lazarus404, 2018-06-08:
Working on this now, but can I confirm; a page will have one of these two strings on it to denote the page number.  So, for example, a page may start with this:

------------------------#34

As such, on the Go To Print Page Number page, if a user enters 34, it should go to that page.  Is this correct?  I am implementing as this now, but can alter it if I read this wrong.

pachpict, 2018-06-11:
In short, yes, that is correct. 

But not every page will have a print page number. It will be something like one in ten pages, for example (a Canute page is usually a third of a BRF page, and a BRF page will be an unknown fraction of a print page). 
Sometimes, in theory, the print page numbers may skip ahead, if a part of the source text is considered irrelevant or untranscribable.

Lazarus404, 2018-06-11:
If it does skip ahead, say from 8 to 12, but the reader enters 10, what should happen?

Lazarus404, 2018-06-11:
Also, do the Braille page numbering use the same numbering format in Go To Page (ie. #A, #JH etc.) or numerical?

Lazarus404, 2018-07-28:
Need to discuss this again as need to know the edge case requirements.  Let's schedule a chat about this when we meet up.

pachpict, 2019-04-01:
Still a popular feature request. Should look at implementing it asap after initial release.