Bristol-Braille/canute-ui #145: Put all menu text into translatable files

kasbah, 2018-02-16:


Lazarus404, 2018-07-16:
So, I have a branch for this.  I've included a dynamic I18n class, so text can theoretically be changed at runtime.  However, I need to modify some modules to make this happen.

Currently, to make my branch change locale, you set an env var `LANGUAGE` to that locale, such as:

```
LANGUAGE=de_DE ./canute_ui
```

I will be changing this to use a config file, once I'm able to get runtime changes working across the entire app.

kasbah, 2018-07-16:
Thanks, and sorry I haven't merged the other PRs yet. I haven't had the opportunity to test on a real canute for quite a while. 

I think in the end this actually needs to be _user_ configurable so should go in a menu rather than a config. I should go into [`state['app']['user']`](https://github.com/Bristol-Braille/canute-ui/blob/6c4d205b6b42445ecc3d4096eadc82773b252905/ui/initial_state.py#L21) and can then be saved into the `canute_state.txt` onto the user's SD-card. I guess that's kind of a config file as well.

Lazarus404, 2018-07-16:
Thanks.  That's what I'm looking into now, so that info is valuable.  I will see what I can get working and will submit a PR when ready.

Lazarus404, 2018-07-25:
Okay, so, I figured it out.  The only bit stopping me now is being able to copy the frozen ordered dict.  The manual key is dyanmic, so not sure how to pass the manual file name as a key in a copy.

```
new_books = books.copy(manual_filename=manual)
```

If I can figure that out, then it's all down hill from here.

Lazarus404, 2018-07-26:
So, my latest push now works from the language select menu.  Tonight, I'll make the system menu dynamic, so that can translate, too.  Then, I'll look to making the translations paragraph based, rather than line based, using a line break helper function.

Lazarus404, 2018-07-30:
So, using Fill seems to screw up the i18n parsing into the POT template completely.  It just skips them.  I spent the better part of yesterday playing with this.  I have some ideas and will see what I can get working, today.

So close!

Lazarus404, 2018-07-30:
So I got it working, but we're gonna have to follow KISS.  If we use the TextWrap module, it doesn't parse to the i18n get_text format, which is a pain.  I'll take a look at the rest of your comment, now, and implement them.

PR #162 (2018-09-06): WIP: Put all menu text into translatable files
Closes #145
closed 2018-09-06 by kasbah