Bristol-Braille/canute-ui #146: Move 'go to start of book' and 'go to end of book' into the bookmarks menu

pachpict, 2018-04-09:
Issue: It is very easy to accidentally press 'go to start/end of book whilst reading, which then loses the reader their place. This has been demonstrated in user trials.

Therefore remove them from the main navigation menu. 

In the bookmarks sub menu add two 'pseudo-bookmarks', i.e. not deletable or defined by the user, labelled "start of book" and "end of book" (whereas other bookmarks are labelled with text from the page, these have fixed labels). 

The former to always be the first bookmark (even if the user defines a bookmark at the first page); the latter to always be the last bookmark in the list, no matter how many user defined bookmarks are added.

pachpict, 2018-05-08:
'Go to end of page' is replicated twice in the book-marks menu with the current implementation.

kasbah, 2018-05-08:
You mean "end of book"?  

I can't reproduce with the emulator. Is it only with a particular book?

 Is it possible you have a book with a bookmark already on the last page?  The new version won't allow making bookmarks there but the old version did. If so, edit the .txt file and remove that bookmark (or just remove the .txt file).

Lazarus404, 2018-07-28:
Any further thoughts on this PR?

kasbah, 2018-07-30:
There seems to have been some confusion around 1-page books (which don't get end-of-book bookmarks). Closing this unless someone can show me missing end-of-book marks on books with more than one page.

PR #150 (2018-05-03): Use permanent bookmarks for start & end of book
fixes #146
closed 2018-07-30 by kasbah