Bristol-Braille/canute-ui #154: System menu says its page one of two, but its only one page.

pachpict, 2018-05-17:
Behaviour observed in March at CSUN, not looked for it since...

kasbah, 2018-05-18:
Page number display code was changed since then so it's unlikely that this is still current behaviour.

kasbah, 2018-05-18:
Oops, nevermind. It does still seem to be doing that.

steph-sergeant, 2018-05-18:
Hi Josh, 

 

Room for further development if it still has 2 pages. 

 

Are you going to put the sort books into latest read first? 

Makes a difference with a large library. 

 

Have a good weekend.

Steph 

 

Stephanie Sergeant 

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email:  <mailto:Stephanie@screenreader.net> Stephanie@screenreader.net 

Website:  <http://www.visionthroughsound.co.uk/> http://www.visionthroughsound.co.uk

kasbah, 2018-05-20:
Hi Steph, who is Josh?  Latest read first would make sense but that's a separate issue. I'll look into this one.

Lazarus404, 2018-07-28:
Any further thoughts on this PR?

kasbah, 2018-07-28:
You are commenting on the issue, not a pull-request. The pull-request you opened for this issue is #161 where I asked you to review my changes. I have requested reviews through the github interface for other pull-requests as well which I don't think you noticed.

Lazarus404, 2018-07-28:
Hey Kaspar,

Yes, I noticed.  I acted on several of the items but haven't had chance 
to complete them all, yet.  I will be tidying it up and rounding it off 
tomorrow, most probably.

I didn't notice posts on the other PR's.  I'll check on them once 145 is 
squared away.

Lee

PR #161 (2018-07-30): Reduce max_pages for system page
Max page count now correct on system page, closes #154
closed 2018-07-30 by kasbah