Bristol-Braille/canute-ui #163: Error I handling page breaks in brfs

pachpict, 2018-06-22:
When reading through mozart score (seafile/canute-books/mk13) canute goes one more line out of synch for every page turn. James Bowden, who produced it, and knows his stuff, Nsists that it has been correctly formatted for a page break every nine lines.

This suggests we are not always handling standard brf page breaks correctly.

pachpict, 2018-11-12:
This has been confirmed  by a calendar formatted in Duxbury for us by George Bell. The file in question is Seafile/canute_books/calendar_2019.brf

[calendar_2019.txt](https://github.com/Bristol-Braille/canute-ui/files/2573044/calendar_2019.txt)

pachpict, 2018-11-12:
It relates to /ui/book/handlers.py, specifically where the form feed is interpreted;

`FORM_FEED = re.compile('\f')`

`           for line in file_contents.splitlines():
                if FORM_FEED.match(line):
                    # pad up to the next page
                    while len(page) < book.height:
                        page.append('')
                    if line == '\f':
                        continue
                    else:
                        line = line.replace('\f', '')
                if len(page) == book.height:
                    pages.append(tuple(page))
                    page = []
                page.append(braille.from_ascii(line))
                if background:
await asyncio.sleep(0)`

We must be handling the replacement of form feeds wrong. Note that the sample file in the previous comment has the correct number of lines per page (9), but that it is apparently standard for the form feed to go at the beginning of the line *afterwards*. So the first form feed is at the beginning of the first line of page 2.

zwbrbr, 2019-05-01:
Fixed in 7d504eb.

closed 2019-05-01 by zwbrbr