Bristol-Braille/canute-ui #164: Diagnostics test book baked into the library

pachpict, 2018-07-17:
Just like the Canute, there should be a Canute test book permanently in the library. This can be various pages of all one character.

steph-sergeant, 2018-07-17:
Very good idea. 

Perhaps a page with the “for” sign all 6 fots in one cell would be good. 

To show any ghost dots a blank page would be useful. 

 

 

Stephanie Sergeant 

 

Vision Through Sound CIC

Tel: 0121 445 2161 

Mobile:  07793 906 745 

Email:  <mailto:stephanie@screenreader.co.uk> stephanie@screenreader.co.uk

For talking books, podcasts, newspapers, phonecalls and texts:

Website:  <http://www.inyourpocket.net/> http://www.inyourpocket.net

CANUTE The First multi line ebook reader in the world

http://www.bristolbraile.co.uk

pachpict, 2019-04-01:
Note comment in #157;
This file to be called "Canute test and clean" or somesuch,

pachpict, 2019-04-01:
This is something I can provide as a BRF.

zwbrbr, 2019-04-15:
Added.  As always the cosmetic aspects are likely to be the sticking points:
 * I'd imagine we'd want this to always appear last in the library menu, but wanted to ask before bothering with that sorting hack.  If I'm doing that, maybe the manual should enjoy the same special treatment.
 * I made the title "for cleaning and testing canute" (a word-swap away from the BRF name), but maybe you really did want "canute test and clean" per the comment above?  Personally I'd mention cleaning first, because in an ideal world of error-free Canutes no testing would be necessary and users shouldn't have to care about it.

If you'd care to opine, @pachpict, I'll tweak as necessary.

zwbrbr, 2019-04-23:
Sorting: @pachpict wants both baked-in books to appear last in the library.

pachpict, 2019-04-30:
And I agree with the name you chose.

zwbrbr, 2019-05-01:
Fixed in dbf2b92 (except sorting, split off into #207).

closed 2019-05-01 by zwbrbr