Bristol-Braille/canute-ui #165: Check BRF compatibility with certain characters

pachpict, 2018-07-17:
Here is a comment from Josh Kennedy on the Braillists Forum, who uses current problems with the BrailleMe as a warning for us;

>     The five problem characters. Dot 4, dots 4-5, dots 2-4-6, dots
>     1-2-4-5-6, and dots 1-2-5-6. For the five dot combinations listed above, it currently
>     expects to see: @^[]\
>     Braille me expects those. But if it sees, `~{}| it just shows empty cells.

In other words, both those sets of characters are valid in BRF, so we should support them all. Perhaps we already do, in which case the issue can just be closed.

pachpict, 2018-08-16:
Follow on comment from James Bowden (RNIB, UKAAF, ICEB) on same forum;

```
I hope you don't mind me picking up a couple of points. I trust this is
helpful:

1. Character set: (Josh's issue 2 and 3).
The problem here is that we have 64 braille dot combinations and 95
printable ASCII values to choose from.
Traditionally, the ASCII values assigned to braille characters were
0x20-0x5f (32-95).

0x40-0x5f are basically the capital letters and a few punctuation signs.
Therefore, in an eight-dot situation, an extra dot 7 showing the capital was
added to all these characters.

If a braille display user has an eight-dot display and is attempting to read
a six-dot file, the extra dot 7s can detract from the reading experience.

Fortunately, lowercase letters, 0x60-0x7e) are also available, so some
braille translation software has adopted these instead, leaving a gap in the
middle.

The NABCC defines 0x60-0x7e as exactly the same as 0x40-0x5e, but without
the extra dot 7. Incidentally, the same is true for the UK Computer Code.

Josh is right to say that the rendering software should have all 95
printable characters defined.

The 96th, would be 0x7f, 127, which is the Delete character. It could be
assigned a printable value equivalent to 0x5f (95), but should not occur in
files.

```

pachpict, 2018-11-08:
In Sight Village many "ou" symbols were shown to be not coming up correctly. Here is a fix from James Bowden (copied out verbatim from quick note in SV London hall, excuse terseness):

`
64 brl
95 ascii
Duplication

Brl - ASCII
0x20-0x5f &
0x60-0x7e
are duplicates. 
Put them both in the Canute BRL table.
`

zwbrbr, 2019-04-08:
Treating this as a dupe of #178

closed 2019-04-08 by zwbrbr