Bristol-Braille/canute-ui #170: Support non US BRF character encodings

pachpict, 2018-08-16:
Different countries use different BRF assignations for Braille characters. 

For our initial launch the US version of BRF, NABCC, is fine (it is supported across the English speaking world). 

However for going into non-English markets we will need to add support for alternative BRF character encodings.

As far as we know there is no way of telling from reading the file which character encoding it is written in, so the only sensible option for our first implementation of this feature is a system-wide menu option (in the system menu) to switch *all* files over to an alternative BRF character encoding.

This will obviously have implications for the cached versions of these files which will need to be considered. 

It will also be cumbersome if people want some BRFs in one character encoding and others in another, on the same Canute. The solution to that would be book-specific switching, but lets treat that as an unusual case and not do that until after system-wide switching has been implemented.

This should come after issue #145 (switch languages in menus) and after the initial release. It does not apply to PEFs.

Here is the comment from James Bowden on the Braillists Forum;

```

> So far, we have assumed the NABCC, which is the character encoding used in
> English-speaking countries. Have you considered other character sets? For
> example, if I obtained a braille file (.BRF) from say, France or Germany, it
> would be encoded with a different character set. For example, in NABCC,
> braille dots 25 is represented by the character 0x33 (51), digit 3; but the
> French and the Euro computer code use 0x5a, 58, colon. The character 0x33 in
> the French and Euro character sets represents braille dots 146. If I can use
> an analogy, it is a bit like having "braille code pages".
> 
> If Canute is going to be used in non-English countries, it will be important
> to have a localisation function so that braille files will display
> correctly.
> 

```

pachpict, 2019-04-01:
This will become important once the Canute has already launched in the UK/Ire/Can/USA (and probably set through either s system menu or a settings file on the SD card).

zwbrbr, 2019-04-08:
In an e-mail thread 2019-02-15 James also pointed out:

> though everything is handled in the USA set in the range 0x20-0x7e, in other country codes you may well need to define up to 0xff, as [octets that appear as accented characters in some encodings] are used for some dot combinations.

pachpict, 2023-09-27:
This requires two settings, each specifying an encoding by name. 

One global and stored in state file, which is default encoding to assume when opening a new book.

Another local to a book and stored in the canute.bookname.txt file.

Former changable by new item in system menu. Latter changable by new item in main menu.

We should start with two baked in encodings. US-NABCC and EuroBraille.

pachpict, 2023-09-27:
Note that the global setting for BRF encoding should also change the file name encoding in the library menu (which is the only dynamically transcribed part of the UI and is done in NABCC right now).

pachpict, 2024-01-17:
Bumped by recent requests for this feature. First implementation should be of a system wide setting. This will be easy enough for a reader to change on a per book basis initially, then once successfully implemented we should introduce a per-book setting stored in the books' 'canute.[bookname].txt' file.