Bristol-Braille/canute-ui #171: Dialogue box: Warnings for when a file exceeds 40 cells wide

pachpict, 2018-08-16:
Some software doesn't wrap BRFs correctly, and a very rare few will have more than 40 cells per line even after wrapping. At present these extra characters are just ignored. We should stick to this approach, but warn the user that is what is happening.

This could be done through a type of dialogue box (which we do not currently have): a simple 'alert' which pops up every time that file is loaded by the user.

To signify difference from a usual page the edge cells would be six raised dots all the way around.

Something like, "Alert: this file has characters going over 40 cells, which are not being shown. Please format the file to 40 cells or less for use on the Canute. Press any button to continue." in the centre of the display.

Then it goes back to the last page the user was on.

kasbah, 2018-08-16:
I don't quite understand why we are so resistant to re-flowing text. Python's built in [textwrap](https://docs.python.org/3.6/library/textwrap.html) module would make it fairly straight-forward to do.

pachpict, 2018-10-08:
For this release of the Canute, at the least, we are prioritising keeping page formatting *exactly* the same as it is for paper Braille. "Anything you would emboss, you can have just the same on the Canute," is the line. This means lines can't be arbitrarily added to, as that would break page layouts. Any file that is over 40 cells wide is almost certainly incorrectly transcribed in any case, so its reasonable to treat it as bad input.*

* The exception being the are 42 cell wide file format. But that would make for dreadful line wrapping (one word on every other line). So we can ignore 42 cell files.