Bristol-Braille/canute-ui #182: Repetitive book loading at start-up

zwbrbr, 2019-03-20:
UI logs indicate that books seem to be loaded multiple times.  Example log fragment from one startup:

```
2018-10-18 08:42:04,000 - ui.book.handlers - INFO     - background loading /media/sd-card/00 Harry Potter and the Philosophers Stone.brf
2018-10-18 08:42:04,077 - ui.book.handlers - INFO     - background loading /media/sd-card/01 Alice_in_wonderland_and_looking_glass.brf
2018-10-18 08:42:04,191 - ui.book.handlers - INFO     - background loading /media/sd-card/02 Pascals Triangle.brf
2018-10-18 08:42:04,286 - ui.book.handlers - INFO     - background loading /media/sd-card/03 Cricket Table.brf
2018-10-18 08:42:04,377 - ui.book.handlers - INFO     - background loading /media/sd-card/04 Calendar March 2018.brf
2018-10-18 08:42:04,469 - ui.book.handlers - INFO     - background loading /media/sd-card/05 population data.brf
2018-10-18 08:42:25,702 - ui.book.handlers - INFO     - background loading /media/sd-card/17_1984.brf
2018-10-18 08:42:25,788 - ui.book.handlers - INFO     - background loading /media/sd-card/06 Mozart Sn14 III.brf
2018-10-18 08:42:25,872 - ui.book.handlers - INFO     - background loading /media/sd-card/07_German_Bentele_Chapter.brf
2018-10-18 08:42:25,908 - ui.book.handlers - INFO     - background loading /media/sd-card/08_diagrams_and_shapes.pef
2018-10-18 08:42:26,438 - ui.book.handlers - INFO     - background loading /media/sd-card/05 population data.brf
2018-10-18 08:42:26,483 - ui.book.handlers - INFO     - background loading /media/sd-card/03 Cricket Table.brf
2018-10-18 08:42:26,582 - ui.book.handlers - INFO     - background loading /media/sd-card/04 Calendar March 2018.brf
2018-10-18 08:42:26,728 - ui.book.handlers - INFO     - background loading /media/sd-card/06 Mozart Sn14 III.brf
2018-10-18 08:42:27,184 - ui.book.handlers - INFO     - background loading /media/sd-card/05 population data.brf
2018-10-18 08:42:27,258 - ui.book.handlers - INFO     - background loading /media/sd-card/06 Mozart Sn14 III.brf
2018-10-18 08:42:27,333 - ui.book.handlers - INFO     - background loading /media/sd-card/04 Calendar March 2018.brf
```

Fix is inbound.

kasbah, 2019-03-20:
Hey Zak, does indeed look like a bit of wasted work there. Should we 
meet to help get you up to speed on UI this week?

zwbrbr, 2019-03-20:
Replied by e-mail.

zwbrbr, 2019-05-01:
Fixed by e77235a.

closed 2019-05-01 by zwbrbr