Bristol-Braille/canute-ui #192: Some BRF books crash or do not show correctly

pachpict, 2019-04-07:
From Joseph Hodge, from APH, referring to 'The Killing Game'  (will be added to Seafile/canute_books/BOOKS_THAT_CRASH/;

> Here you go here is a BRF I tested with that gave us a lot of blank pages
> and things I didn't see when using a normal computer with a display.

I have requested information about the source of the file and will append any response to this issue as a comment.

pachpict, 2019-04-09:
It came from Bookshare, an internet BRF library, primarily American, which hosts books generated from a variety of sources. Added to Seafile.

zwbrbr, 2019-04-10:
Can't reproduce this in the Qt UI; at least, I see no blank pages as I scroll through the book.  Won't close it before trying on real hardware, though.

"things I didn't see when using a normal computer with a display" --- it's hard to rule out seeing differences between the Canute and another medium without more specifics!

We'd need to get down into details to proceed with this one; in particular:
 * Which page numbers are blank on the Canute but have content via another medium?
 * What specific Braille cells are added or different on the Canute?

zwbrbr, 2019-04-26:
I note that "The Killing Game" is a weighty tome relative to the books we usually test with.  I think I've seen placeholders for page numbers and blank pages _until it loads_, which takes a while on real hardware purely because it's big.  From memory, background loading doesn't incrementally update the manifestation of the book; it just keeps the UI responsive while loading, then swaps the book into place (so book presence is all-or-nothing).

Not sure how #182 might interact here (I forget whether duplicate loads would manifest visibly, beyond causing lag), but one could speculate that #182 plus a large book in a large library might require a surprisingly long delay between power-on and settled, readable books.

zwbrbr, 2019-09-17:
@pachpict Info never appeared so we have no example of any breaky book; suggest we close this until we do.

zwbrbr, 2019-09-25:
Closing and we'll open new ones for specific broken books.

closed 2019-09-25 by zwbrbr