Bristol-Braille/canute-ui #195: Manual's last page mis-renders

zwbrbr, 2019-04-08:
View the manual, step through the pages.  Penultimate page:

```
1           system menu
2
3 Configure your preference on the sorting
4 order of books in the library and
5 bookmarks through the menu options. To
6 shutdown the Canute safely, select the
7 shutdown option and wait for #cj
8 seconds before unplugging it.
9
```

Hit next, and what gets left on the display is:

```
1
2
3
4
5
6 shutdown the Canute safely, select the
7 shutdown option and wait for #cj
8 seconds before unplugging it.
9
```

Seen in the Qt UI, for lack of dev hardware; may or may not affect real hardware, but if it doesn't there's a simulation inaccuracy bug to be raised.

I suspect this is because the baking-in of the manual leaves it in a non-canonical format, specifically without padding the final page out with blank lines.

pachpict, 2019-08-20:
Is this issue still unresolved?

zwbrbr, 2019-09-17:
Fixed in a roundabout way in 9b969a9.  Baked-in books will go away eventually anyway, but in the meantime if they change we just need to check for this.

closed 2019-09-17 by zwbrbr