Bristol-Braille/canute-ui #198: Support language packs on external media

zwbrbr, 2019-04-25:
Split off from #124.

@pachpict Still needs clarification on what level of expertise we expect a translation producer to have.  Producing the translations is one part of it, and I expect the answer is "able to use Transifex".  The other part is getting those translations into the compiled format expected by i18n libraries and putting them in the correct directory structure.  So, are we leaving translators to navigate that or are we asking them to contribute translations back to us, which we package for them plus integrate into the next release?

zwbrbr, 2019-04-25:
If a locale dir on media follows conventions, we can employ normal resolution with `gettext.find()` on additional dirs, and open a found `.mo` with `gettext.GNUTranslations(file_obj)`.  Other parts of the puzzle are enumerating available translations in a sensible way, and a way to put the translation's menu item text into, and retrieve it from, `.mo` files (probably in the MIME-esque metadata section).

pachpict, 2019-08-20:
For now we can get away with adding language through UI updates.

woodcoder, 2025-01-05:
Updateed notes: It should prefer ones on external media over ones loaded internally if the names clash.  It will need to remember them, but it doesn't need to be very intelligent and it can forget preference and revert to UEB G2 when the media is removed, even if the media is later restored.

PR #316 (2025-01-05): Allow translations to be provided on external media
This PR provides support for extending or overriding the standard gettext language translations for the canute-ui.

The change will look for a `locale` directory on removable media (SD card or USB drive) containing files such as: `de_DE.UTF-8@ueb1/LC_MESSAGES/canute.mo`

These should be compiled from the `canute.po` file containing translations, based on the most up to date `canute.pot` template file.

When found, each file is checked for a `msgid "Language Name, UEB grade"` entry which should provide the actual language name for that file, i [... cut]
closed 2025-01-05 by woodcoder