Bristol-Braille/canute-ui #203: Allow Canute to be controlled as a display again

pachpict, 2019-04-30:
Allow Canute to  be controlled as a display through the USB-B port, especially by screen readers such as NVDA. (Note that this was previously supported by the Canute UI and support was built into a build of Chrome OS, but has been subsequently disabled due to changes to the UI and in the underlying hardware.)

zwbrbr, 2019-05-14:
Making this dependent upon [firmware issue #17](https://gitlab.com/Bristol-Braille/canute-firmware/issues/17) because we don't want to spread/inflict the protocol sync/async inconsistency wider than necessary.

zwbrbr, 2019-09-17:
USB was re-enabled as of `pi-misc` `90a936a`.  Protocol could certainly be improved, but we *could* ship it as-is.

zwbrbr, 2019-09-25:
Next step: try using UI natively with external Canute on USB, just to see.

zwbrbr, 2019-09-26:
UI runs with an external Canute, which is good enough for now.  Can raise separate bugs for protocol concerns, albeit against Gitlab firmware repo rather than here.

closed 2019-09-26 by zwbrbr