Bristol-Braille/canute-ui #212: Reader should be able to delete a bookmark

pachpict, 2019-05-01:
Can we already do this? Forgive me if I should know the answer to this, as I've forgotten. @kasbah and @zwbrbr your input welcome. Note to self: this is in reference to APH Redmine issue 5596.

zwbrbr, 2019-05-05:
@pachpict From the code (not tried it) a long-press on a bookmark deletes it; this seems to have escaped documentation.

zwbrbr, 2019-05-08:
Long-press works, and is documented in the context help on the APH units; they must just have missed it.  There's a [note](https://github.com/Bristol-Braille/canute-ui/commit/564e49d105586376362292466d43b3e4a8a83035#diff-f8836d55d69518a8d2f51c13010c5b06L14) to ensure it stays in the context help.

@pachpict I guess you can close Redmine 5596 with this info?

zwbrbr, 2019-09-24:
@liamsmyth92 Per my comment May 8th, when Ed and I reviewed your context help updates in May we [made a review note to you](https://github.com/Bristol-Braille/canute-ui/commit/564e49d105586376362292466d43b3e4a8a83035#diff-f8836d55d69518a8d2f51c13010c5b06L14) pointing out that you'd dropped the bookmark deletion method, among the other suggested fixes.  I didn't re-check your amended text when you re-sent it; I just assumed that you'd addressed all problems we'd raised.

Simply revise [the existing text](https://github.com/Bristol-Braille/canute-ui/blob/master/ui/bookmarks/help.py#L8) and get your revised version to me, and I'll pop in for the next release.

liamsmyth92, 2019-09-26:
Sent to you, added the delete bookmarks text back in.

commit 99cbddc0 (2019-10-10): Document bookmark deletion in help

Fixes #212.
closed 2019-10-10 by zwbrbr