Bristol-Braille/canute-ui #243: No support for XFAT on USB sticks and SD cards?

pachpict, 2019-08-20:
Not high priority for first release but will cause unnecessary grief for support staff. Meanwhile we will upgrade the manual.

I wasn't sure exactly how to read @timburg1's observations on XFAT compatibility. 

Test 10 implied there was XFAT support:
https://github.com/Bristol-Braille/canute-ui/wiki/Testing-report-by-TB-from-2019-07-24-to-2019-07-30#test-10

While test 14 suggests it works, but takes a minute before it shows up:
https://github.com/Bristol-Braille/canute-ui/wiki/Testing-report-by-TB-from-2019-07-24-to-2019-07-30#test-14

Can you clarify Tim?

pachpict, 2019-08-29:
I'm moving this to being for the first consumer release as it is hopefully a small fix.

zwbrbr, 2019-09-25:
Support is present, mounting of exFAT has been confirmed working, so it remains to be seen why/when this is failing.