Bristol-Braille/canute-ui #250: Support sorting books into directories to make the Library more managable

pachpict, 2019-08-20:
https://github.com/Bristol-Braille/canute-ui/wiki/Testing-report-by-TB-from-2019-07-24-to-2019-07-30#conclusion

Also noted by a variety of testers over the years. We are going to have to bite the bullet and add it in after the initial release. 
Will show directories on the SD card/USB stick as ending with a forward slash (truncated if needs be), as per the usual convention.
When pressing that button go into a library menu which says something like "Library Menu: /usb/myfolder/.....#1/#3"
Then when in a sub directory the first entry (next to button 2) is always "../ Go back to parent directory", not only on the first page of the sub-directory's list but on every page.
Pressing the Library Menu from an open book should go back to the last place in the directory structure, which will need to be recorded in the state file.

pachpict, 2021-11-30:
TVision have confirmed that absence of this feature has lost at least one customer. Specifically important is the inability to read through the SD card given out by RNIB library, especially to Orbit Reader 20 owners. 

We knew this already, but is useful to record real-life consequences of not implementing it.

pachpict, 2022-02-11:
As first release does not necessarily need to support more than 128 or 256 books per directory, whatever helps avoid necessity for speed or UI optimisations.
Could use existing Rust optimised book loading structure (see #253 for some future improvements to be made to this structure), or could scrap that if optimisations can be circumvented. The letter is probably a much better route.

pachpict, 2022-03-22:
Simplest method (to try out at least, if not as final implementation) may be to show as a tree structure rather than a Norton/explorer style pane, so every directory is shown, but only files within the current directory are shown.

 1. [Modify `def title` in `book_file.py`](https://github.com/Bristol-Braille/canute-ui/blob/master/ui/book/book_file.py) so it returns the full path.

Would look something like:

``
    Library menu ---------------------------------------- 1/8
    /media/sd-card/sd-test-file
    /media/sd-card/RNIB Library/Adult Fiction/P/Pepper Pig - Return to Helmand - Vol1 G2
    /media/sd-card/RNIB Library/Adult Fiction/P/Pepper Pig - Return to Helmand - Vol2 G2
    /media/sd-card/RNIB Library/Adult Fiction/Readme
    /media/sd-card/RNIB Library/Adult Fiction/Index of adult fiction titles
    /media/usb-stick/usb-test-file
    ... etc...
    ... etc...
``

 3. Add 'current browsing directory' (CBD) to the state (defaulting to SD card root).

 4. Remove '/media/'. Truncate 'sd-card/' and 'usb-stick/' to 'sd/' and 'usb/'. Truncate all directory names that are not the current deepest level of the CBD path. Like:

``
    Library menu ---------------------------------------- 1/8
   sd/sd-test-file
   sd/RNI.../Adu.../P/Pepper Pig - Return to Helmand - Vol1 G2
   sd/RNI.../Adu.../P/Pepper Pig - Return to Helmand - Vol1 G2
   sd/RNI.../Adu.../Readme
   sd/RNI.../Adu.../Index of adult fiction titles
    usb/usb-test-file
    ... etc...
    ... etc...
``

 5. Only return file names for files in the CBD and then get rid of duplicate entries. Like:

``
    Library menu ---------------------------------------- 1/8
   sd/
   sd/RNI.../Adu.../P/Pepper Pig - Return to Helmand - Vol1 G2
   sd/RNI.../Adu.../P/Pepper Pig - Return to Helmand - Vol1 G2
   sd/RNI.../Adu.../
    usb/
    ... etc...
    ... etc...
    ... etc...
``

 6. Change [`reducers.py`](https://github.com/Bristol-Braille/canute-ui/blob/master/ui/library/reducers.py) so if a user clicks on a path (rather than a file within CBD) then it *doesn't* load a file, but instead changes the CBD and reloads the library (some how).

And that might work, or at the least will give us a good idea of how to proceed.

pachpict, 2022-03-23:
Above proposal requires supporting `truncate_middle` (in `braille.py`) with another function like `truncate_path`. How does it handle truncating directories within the path? Not at all simple. 

In theory could run out of space and be unable to go to deeply embedded directories. One, dirty, approach to that would be to flat out say we don't support more than eight directories deep, a la ISO9660.

More of an issue is may not have enough space to show the file names.

Approach: Always prioritise the filename and lowest directory name for paths in the CBD, and always prioritise only the last directory name in the path for paths outside the CBD. That way the reader can always read the title of, and navigate to, any directory with books in it, even if the path of that directory is obscured. Obscuring the path is not ideal because, unless there are BRFs in every directory, at every level, some directories may not be represented at all. So for example these two:

``

  /sd-card/RNIB Library/Adult Fiction/P/Pepper Pig - Return to Helmand - Vol1 G2
  /sd-card/RNIB Library/Kids Fiction/P/Pepper Pig - Adventures in Happy Land - Vol1 G1

``

Become:

``

  ////P/Pepper Pig - Return to Helmand - Vol1 G2
  ////P/Pepper Pig - Adventures in Happy Land - Vol1 G1

``

... Losing key context (unless there was at least one BRF in both 'Adult Fiction' or 'Kids Fiction', then it would render those directory as options, separating the two books).

Could take ugly approach of setting maximum space directory names in path can take up ((40/2)-`current_page`); then if directory names in path not within that run `truncate_middle` on each directory name (with a `width` argument of something arbitrary like 8); then if directory names in path still not within that just chop it off at the start with an ellipses.

(If needed: For splitting a string by multiple delineators see: https://stackoverflow.com/questions/4998629/split-string-with-multiple-delimiters-in-python#4998688)

pachpict, 2023-09-05:
Design notes to be implemented:

```

/media/sd-card/RNIB Library/Adult Fiction/P/Pepper Pig - Return to Helmand - Vol1 G2

LIBRARY menu - sd/rni... ry/adult fiction/p/ - #nn/#nn
-front usb/
-sd card/
--rnib library/
---adult fiction/
----p/
pepper pig - return to helmand vol1 g2 - #745/#744
pepper pig - return to helmand vol2 g2 - #866/#866
pepper pig - return to helmand vol3 g2 ---- #1/#nn
----r/
---kids fiction/ 

LIBRARY menu - sd/rni... ry/adult fiction/ - #nn/#nn
----t/
----u/
----v/
----w/
----y/

```

Caching:
Up to 8 at a time, as they appear on the library menu screen and up to 8 either side, as per #253, and #206, also helps with #196, makes #98 unneccesary for now, check if it fixes #205, do *not* do #271.

StephanieSeargent, 2023-09-10:
Hi Andy, 

 

I am Steph who does the user design and  technical support for BBT. 

 

This is thrilling, being able to read directories will make the RNIB card usable. 

 

I am so glad you are doing it! 

 

All the best,

Steph 

 

 

Stephanie Sergeant 

Email  ***@***.***> ***@***.***

Mobile 07793-906-745

WEB  <http://www.visionthroughsound.co.uk> www.visionthroughsound.co.uk

closed 2023-11-08 by woodcoder