Bristol-Braille/canute-ui #275: Apparently possible to permanently block Canute-as-display functionality from working

pachpict, 2022-07-15:
System: Canute Dock, Pi400 running Raspian with [BRLTTY for Canute installed](https://github.com/brltty/brltty/tree/bbt-canute-screen).

Canute: Tested in three different Canute 360s from batch 3. All running stock 1.3.19.

Detail: 

After being used for some time as displays for Debian console, in all three cases being turned on and off in a variety of ways (see below), two stopped acting as displays at all and weren't showing up in `/sys/kernel/debug/usb/devices` at all.

Issue is assumed (without adequate information) to be to do with when the Canute's are turned on and off relative to the PC. For these two was not being specific as didn't realise it would be an issue (was trying to solve another issue to do with XWindow support so restarting frequently). Tried all manner of combinations of turning off and on, including:

 - Canute turned on first then Pi turned on after Canute had booted to book menu.
 - Canute and Pi turned on at same time (in which case Canute's book up is interrupted after first hard refresh, before it gets to the book menu, to act as a display).
 - Pi turned on first, Canute turned on second, then plugged in to Pi after booting to book menu.
 - Pi turned on first, Canute turned on second whilst plugged in to Pi.

At some point both Canutes stopped responding/acting as display and just boot to book menu. They do however tick fast, characteristic of being plugged in to a PC (a minor bug we need to fix).

Tried to replicate this with a third but despite repeating all the combinations above it continues to enter display mode. 

Idea 1: Something has flipped in those two Canutes preventing them from going into display mode. 

Action 1: Should reflash the SD card (as virgin) and try again. But should first look for that 'flag'.

Idea 2: It may have had nothing to do with the sequence of turning off/on.

rbubley, 2022-07-15:
Knowing virtually nothing of the inner workings, there is something about the description of the problem that makes me think of memory corruption coming from power surges.

Or maybe not - just an idea...

mattvenn, 2022-07-16:
One idea to help debug process: Don't reflash the cards, they are evidence
and also possibly the cause.